realsenseai/librealsense #4500: unable to debug

galikrosentalik, 2019-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715| 
| Operating System & Version |  Ubuntu 18 | 
| Kernel Version (Linux Only)    |  4.18.0-25-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.24.0 | 
| Language                            |  C++ | 
| Segment			|  others | 

### Issue Description
Hi everybody, 
Working with the SDK in my system I notice behavior is very different in debug vs release.
In debug, I found the camera output to be invalid. For instance in some cases I get stuck in the "rs2::pipeline.start(rs2::config)" method, and in other cases the "rs2::pipeline.wait_for_frames" method fails.
In release, everything seems to work fine.
Can anyone assist please?

Thanks, Gal.

dorodnic, 2019-07-25:
Are you using GDB? IDE of some sorts?
I never had a problem to run the debug version.

galikrosentalik, 2019-07-25:
I am using clion debug

galikrosentalik, 2019-07-25:
I am using Clion with Cmake and GDB.

I believe somehow the issue is related to libtm state machine goes into timeout:

```
07:48:50.898 [2413] [E] infra/Poller: poll: epoll_wait error 4
07:48:50.898 [2413] [E] Dispatcher: handleEvents(): Poller::poll() ret -1
```

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi @galikrosentalik

This has been open for a while now.  Are you able to debug with latest SDK release?

Thanks

galikrosentalik, 2019-08-24:
Unfortonatly no.
I would appreciate very much you referencing my last comment, reaplaying dorodnic

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

I'm thinking it has to do with system configuration and setup.  We typically use Core i7 based NUCs when we are testing and/or utilizing the tracking camera.

When it comes to timeouts it's usually due to system config or using USB2 connection instead of USB3 connection.

Is there any system specific information you can provide that would help us attempt to replicate the issue?

Thanks

galikrosentalik, 2019-09-18:
It seems i can not make the T265 identify as a usb 3 device. Iv'e tried all ports on my pc using the original cable supplied by the company.
moreover, when i connect D415 or D435 it does identify as usb 3.
can you advise?

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@galikrosentalik - So if you attempt to just run RealSense Viewer (latest version 2.28.0) without any special debug setup or special settings, the T265 device is not identified?

galikrosentalik, 2019-09-19:
I am using RealSense Viewer latest version 2.28.0.
The T265 does identify, however I cannot find the usb connection type.
Please rerfer to the screenshot which shows clearly that the USB connection type is shown with the D435 (USB 3.2) but not with the T265. Please advise how to tell the USB connection type for T265.
 
[image]

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @galikrosentalik

Based on your image that you provide here I can tell you're connected via USB3 since you're able to get the fisheye image streams, as well as the gyro and accel streams along with pose stream, in RealSense Viewer.

Looking at the image, the device is being enumerated and identified along with the D435i that you have connected as well.

I'm not sure how to reproduce your original issue as outlined in your initial post.

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

Hi Gal. Are you still seeing differences between debug and release builds? Do you have #debug specific switches in your code? Would you mind sharing the source so we can see if it's reproducible on a similar setup here?

closed 2019-10-23 by RealSenseCustomerSupport

galikrosentalik, 2020-02-10:
Hi, sorry for the delay.
I still suffering from this issue. can we re-open the issue? 
is there any more info you need from me in order to investigate this issue?

zainmehdi, 2020-02-17:
I am having the same issue with T265 using qtcreator as an IDE. Please provide some insight.