realsenseai/librealsense #4501: movement recognition

teriterance, 2019-07-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 /D435/ T265/ SR300} | | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17/18)} | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.0. }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Good day 
I want to know if any of then camera provide movement recognition algorithm build in. Like hand motion recognition.

MartyG-RealSense, 2019-07-25:
The SR300 has the best support for hand and finger tracking.  A brand new version of the camera called the SR305 went on sale in Intel's online Click store just yesterday.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-sr305.html

One of the Intel guys on this forum such as @dorodnic will be best placed to advise how these features might be supported in the current RealSense SDK 2.0 though.

teriterance, 2019-07-26:
Hi @MartyG-RealSense 
Is it possible to pass filter to the camera. I want to have just a part of image who are distance less than fixed value.

MartyG-RealSense, 2019-07-26:
In the RealSense SDK you can set a 'depth clamp' that sets a maximum distance, and the part of the image beyond that depth is ignored.  You can also reduce the background with a function called a Disparity Shift, that reduces the minimum distance (allowing the camera to get closer to an object) whilst also reducing the maximum observable distance.

closed 2019-08-02 by teriterance