realsenseai/librealsense #4502: Realsense does not open/ No Video

Suri111200, 2019-07-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC (Intel NUC) |
| SDK Version                            | 2.23.0.909 | 
| Language                            |  Python | 

I'm having an issue where I cannot open the Realsense Viewer (the application will not open whatsoever). Furthermore, when running my Python Scripts, a Realsense window opens, however the window is blank and the video is not displaying.

[image]

This all works fine on my other devices, but I'm trying to make it work on this specific PC. Everything is being properly detected and appears correctly int he Device Manager. However, the Realsense application just will not work. How should I go about troubleshooting this next?

[image]

MartyG-RealSense, 2019-07-25:
Problems with opening the RealSense Viewer on a particular computer may sometimes be related to the OpenGL video drivers.  Could you try updating this PC's graphics driver please and see if that makes a difference?  Thanks!

ev-mp, 2019-07-28:
@Suri111200 , are you able to run command-line demos, such as `rs-depth`, `rs-enumerate-devices` and what are the outputs ?

Suri111200, 2019-07-30:
Hey! Just an update on this:

I suspected the issue could be related to my usage of remote desktop, and when connecting a physical display to the machine, everything worked as expected. To further add confusion, it also appeared to work as expected through remote desktop the next day, and I have not experienced any issues since.

While I cannot exactly pinpoint what the issue was, I no longer have an issue. Thanks, everyone!

MartyG-RealSense, 2019-07-30:
Great news!   Thanks so much for the update  :)

closed 2019-07-31 by Suri111200