realsenseai/librealsense #4503: Only able to stream frames when using IDLE otherwise Frames didn't arrive

RoWe90, 2019-07-25:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435/D435i  | 
| Firmware Version                   |  | 
| Operating System & Version |   {Win (10) /                                            | 
| Platform                                 | PC                                                          |
| SDK Version                            |  { legacy / 2.16.1 }                          | 
| Language                            |  python                       | 
| Segment			|  others                         | 

### Issue Description

Hi everyone,

i have a rather annyoing issue:
I can only stream images from cameras if i use IDLE to execute my scripts. I tried several other methods (PyCharm, Windows Command Line Tool, PowerShell...) but i am on none of them able to receive images.

My scripts (or the examples from the GitHub page, doesn't matter) always give out the error **" Frame didn't arrived within 5000"** when the script is trying to execute **pipeline.wait_for_frames()**

Any ideas on how to fix that?

Thanks in advance!

dorodnic, 2019-07-25:
Sounds like you have more then one version of the SDK installed on your system. 2.16.1 was created before D435i existed, so there is no chance of it recognising that camera. 
Please make sure you don't have extra hidden versions of librealsense installed, and upgrade to newer version

RoWe90, 2019-07-25:
I had a look, and couldn't find other versions of the SDK installed. I tried updating both the camera to the newest firmware (5.11.100) and the SDK (2.24.0). I can stream from the RealSense Viewer without any problems. Streaming when starting with IDLE is also possible.
Yet, other methods (PyCharm, cmd-window) do not work.
I just stumbled upon this note at the GitHub page of pyrealsense 2.

**Note:
pyrealsense AKA pyrealsense/2.0 is a community supported Python wrapper for librealsense v1.12.1, This wrapper does not support newer versions and does not work with the RealSense SDK 2.0.**

As far as i understand this, streaming should not work AT ALL, or should it? But why does it work through IDLE then? I am really at a loss here...

RealSenseCustomerSupport, 2019-08-20:
----------------------------------------------

Hi,

Pyrealsense doesn't support the RealSense SDK 2.0, therefore there is no well-defined behavior guaranteed when you use them together.

Thank you!

RealSenseCustomerSupport, 2019-08-27:
----------------------------------------------

Please let us know if you need any further assistance with this matter.

closed 2019-09-03 by RealSenseCustomerSupport