realsenseai/librealsense #4504: Depth accuracy varies across the region/pixels

ShettyHarapanahalli, 2019-07-25:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D415 , D435 &  D435i |
|**Serial Number**|839112061189|
|**Firmware Version**|05.11.06.250|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Does the depth accuracy varies from a location in cameras view, I mean if the ROI is in the first quadrant the accuracy is different compared to ROI in 2nd quadrant?  

Going further if I break cameras view into 15 or more segments will the accuracy of depth in all the segments be same, is there any case study done in the same. If so can you please share the details of it.

dorodnic, 2019-07-26:
As a quick answer - as far as I know, yes, depth quality will vary slightly within the image. 
The reason for that is that lenses and calibration process are not perfect and will introduce slight unaccounted distortions to the two infrared images. This is more apparent when using the D415. I never noticed any such artifact with the D435/i as they pack slightly better lenses.

ShettyHarapanahalli, 2019-07-29:
Good to know that D435i has no such issues, how accurate is d435i compared to d415.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi ShettyHarapanahalli,
From depth accuracy point of view, D415 is about 2x better than D435/D435i. Because D415 uses 2M depth sensor; D435 uses 1M depth sensor. D415 FOV is much smaller than D435.

ShettyHarapanahalli, 2019-08-26:
Thanks for the details :)

closed 2019-08-26 by ShettyHarapanahalli