realsenseai/librealsense #4508: make error trying to build Intel Aero Realsense

RohanKurane, 2019-07-25:
Hello - 

I am following the steps in the Intel Aero Wiki to install librealsense but run into the following error during the `make` step. 
https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#intel-realsense-sdk

```
 sudo apt-get -y install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev cmake
 git clone -b legacy --single-branch https://github.com/IntelRealSense/librealsense.git
 cd librealsense
 mkdir build && cd build
 cmake ../ 
 make     -->  fails as shown below
```

Here is the output from make 

```
make
Scanning dependencies of target realsense
[  1%] Building CXX object CMakeFiles/realsense.dir/src/archive.cpp.o
[  3%] Building CXX object CMakeFiles/realsense.dir/src/context.cpp.o
[  5%] Building CXX object CMakeFiles/realsense.dir/src/device.cpp.o
[  7%] Building CXX object CMakeFiles/realsense.dir/src/ds-device.cpp.o
[  9%] Building CXX object CMakeFiles/realsense.dir/src/ds-private.cpp.o
[ 11%] Building CXX object CMakeFiles/realsense.dir/src/f200.cpp.o
[ 13%] Building CXX object CMakeFiles/realsense.dir/src/hw-monitor.cpp.o
[ 15%] Building CXX object CMakeFiles/realsense.dir/src/image.cpp.o
[ 17%] Building CXX object CMakeFiles/realsense.dir/src/ivcam-private.cpp.o
[ 19%] Building CXX object CMakeFiles/realsense.dir/src/ivcam-device.cpp.o
[ 21%] Building CXX object CMakeFiles/realsense.dir/src/log.cpp.o
[ 23%] Building CXX object CMakeFiles/realsense.dir/src/motion-module.cpp.o
[ 25%] Building CXX object CMakeFiles/realsense.dir/src/r200.cpp.o
[ 26%] Building CXX object CMakeFiles/realsense.dir/src/rs.cpp.o
[ 28%] Building CXX object CMakeFiles/realsense.dir/src/sr300.cpp.o
[ 30%] Building CXX object CMakeFiles/realsense.dir/src/stream.cpp.o
[ 32%] Building CXX object CMakeFiles/realsense.dir/src/sync.cpp.o
[ 34%] Building CXX object CMakeFiles/realsense.dir/src/timestamps.cpp.o
[ 36%] Building CXX object CMakeFiles/realsense.dir/src/types.cpp.o
[ 38%] Building CXX object CMakeFiles/realsense.dir/src/uvc-libuvc.cpp.o
[ 40%] Building CXX object CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o
/home/ubuntu/librealsense/src/uvc-v4l2.cpp: In constructor ‘rsimpl::uvc::subdevice::subdevice(const string&)’:
/home/ubuntu/librealsense/src/uvc-v4l2.cpp:158:39: error: ‘V4L2_CAP_META_CAPTURE’ was not declared in this scope
                 if((cap.device_caps & V4L2_CAP_META_CAPTURE)){
                                       ^
CMakeFiles/realsense.dir/build.make:542: recipe for target 'CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o' failed
make[2]: *** [CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/realsense.dir/all' failed
make[1]: *** [CMakeFiles/realsense.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

The variable `V4L2_CAP_META_CAPTURE` is not defined anywhere in the source code after the git clone. 

Am I missing a command or 2 ?

@dorodnic @sgorsten @aangerma 

Thanks

RohanKurane, 2019-07-26:
If I comment out these 3 lines from src/uvc-v4l2.cpp everything works fine
```
        if((cap.device_caps & V4L2_CAP_META_CAPTURE)){
            is_metastream=true;
        }
```

Where is this variable defined ?

RohanKurane, 2019-07-28:
Any ideas here that can help ?

ev-mp, 2019-07-28:
@RohanKurane , this var definition is part of the linux disto toolchain. It is defined in "usr/include/linux/videodev2.h" kernel header file.
It could be the case that Ubuntu kernel for Aero platform is not up to date, but there is a [fix for this in ther SDK](https://github.com/IntelRealSense/librealsense/blob/master/src/linux/backend-v4l2.h#L48-L67) for more than a year now, and I wonder why it failed in your case.

Can you check whether the above preprocessor code is invoked correctly ?- should produce "Librealsense notification: V4L2_META_FMT_UVC was not defined, adding metadata constructs" message during cmake compilation

ev-mp, 2019-07-28:
@RohanKurane , sorry I misread your post and did notice that you're cloning the `legacy` branch.
It seem though that a [recent contribution to the legacy branch](https://github.com/IntelRealSense/librealsense/commit/7b41b592fba4e1248d218e6abb0f4291e2486bf0) breaks the build (at least) for Aero platform.

We'll need to address it, in the meantime you can add [these lines](https://github.com/IntelRealSense/librealsense/blob/master/src/linux/backend-v4l2.h#L59-L61) to your code as a workaround.

RohanKurane, 2019-07-28:
@ev-mp Thanks. 

When you say add these lines to my code - 
```
#ifndef V4L2_CAP_META_CAPTURE
#define V4L2_CAP_META_CAPTURE    0x00800000  /* Specified in kernel header v4.16 */
#endif // V4L2_CAP_META_CAPTURE
```

where do you mean exactly ?  Right now I basically modify the linux/videodev2.h to add the `#define` of the variable and the `cmake` and `make` work. Adding it to my application which does a git clone and then `cmake` and `make` of it (independent of my application) would not work. I would like to automate this whole process without having to change any code outside of my application.

Any idea when this will be fixed in the legacy branch so It works out of the box with the `git clone -b legacy` ?

ev-mp, 2019-07-29:
You can put these line in the beginning of the affected `uvc-v4l2.cpp`.

Regarding the fix - we'll try to push it in either the coming or the next releases. In the meanwhile you may choose between apply the fix locally or reverting the `legacy` branch to one of the recommended/verified tag - e.g. `v1.12.1`

RohanKurane, 2019-07-29:
Thanks @ev-mp ! I will wait for your fix. 
In the meantime, can you send me the exact `git clone` command to clone the `v1.12.1 legacy branch` for Intel Aero ?

ev-mp, 2019-07-29:
Once you've instantiated the git repo use
>git fetch --all -f -t
git checkout tags/v1.12.1 -b my_local_branch

RohanKurane, 2019-07-29:
Great! so here are my commands. Let me know if I am wrong
``
ke
    git clone -b legacy --single-branch https://github.com/IntelRealSense/librealsense.git
    git fetch --all -f -t
    git checkout tags/v1.12.1 -b my_local_branch
    cd librealsense

    mkdir build && cd build

    cmake ../
    make
    sudo make install

```

ev-mp, 2019-07-29:
There are couple of redundant options ( --all) , but otherwise it should be ok.
Two notes - you're not passing flags to cmake, so the default is Debug mode and I'm also not sure whether the demos are opt in/out. Also it would make sense to adjust the `my_local_branch` name which I used as a placeholder

RohanKurane, 2019-07-29:
Debug mode is fine. Here what I am planning to run - 
``
 git clone -b legacy --single-branch https://github.com/IntelRealSense/librealsense.git
    git fetch --all -f -t
    git checkout tags/v1.12.1 -b my_local_branch
    cd my_local_branch

    mkdir build && cd build

    cmake ../
    make
    sudo make install
```

RohanKurane, 2019-07-30:
That seems to work. Thanks @ev-mp 
I will keep an eye out for the fix to go in

closed 2019-07-30 by RohanKurane

ev-mp, 2019-08-27:
@RohanKurane , the fix is merged into the `legacy` branch

PR #4738 (2019-08-27): Fix legacy build for kernels prior to v4.16
Fixes #4508