realsenseai/librealsense #4510: oneshot record image error

sailfish009, 2019-07-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10 64bit  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.24.0                   | 
| Language                            | C++                       | 
| Segment			|  VR/AR/others                       | 

### Issue Description
i have modified librealsense code like below to record image only 1 frame:
once recorded, i opened bag file with realsense viewer, sometimes it fine, but sometimes it shows like this(which is color image is just white), how i can get fine (always!) 1 frame recorded bag file with librealsense? 

[image]

```
// record 1 frame 
void librealsense::record_device::write_data_once(size_t sensor_index, librealsense::frame_holder frame, std::function<void(std::string const&)> on_error)
{
    //write_data is called from the sensors, when the live sensor raises a frame

    LOG_DEBUG("write frame " << (frame ? std::to_string(frame.frame->get_frame_number()) : "") <<  " from sensor " << sensor_index);

    std::call_once(m_first_call_flag, [this]()
    {
        initialize_recording();
    });

    //TODO: restore: uint64_t data_size = frame.frame->get_frame_data_size();
    uint64_t cached_data_size = m_cached_data_size; //TODO: restore: (+ data_size)
    if (cached_data_size > MAX_CACHED_DATA_SIZE)
    {
        LOG_WARNING("Recorder reached maximum cache size, frame dropped");
        on_error("Recorder reached maximum cache size, frame dropped");
        return;
    }

    m_cached_data_size = cached_data_size;
    auto capture_time = get_capture_time();
    //TODO: remove usage of shared pointer when frame_holder is copyable
    auto frame_holder_ptr = std::make_shared<frame_holder>();
    *frame_holder_ptr = std::move(frame);
    (*m_write_thread)->invoke([this, frame_holder_ptr, sensor_index, capture_time/*, data_size*/, on_error](dispatcher::cancellable_timer t) {
        if (m_is_recording == false)
        {
            return; //Recording is paused
        }
		int device_count = 0;
		std::call_once(m_first_frame_flag, [&]()
		{
			try
			{
				write_header();
			}
			catch (const std::exception& e)
			{
				LOG_ERROR("Failed to write header. " << e.what());
				on_error(to_string() << "Failed to write header. " << e.what());
			}
		});

		switch (sensor_index)
		{
		case 0:
			std::call_once(m_first_color_flag, [&]()
			{
				try
				{
					const uint32_t device_index = 0;
					auto stream_type = frame_holder_ptr->frame->get_stream()->get_stream_type();
					auto stream_index = static_cast<uint32_t>(frame_holder_ptr->frame->get_stream()->get_stream_index());
					m_ros_writer->write_frame({ device_index, static_cast<uint32_t>(sensor_index), stream_type, stream_index }, capture_time, std::move(*frame_holder_ptr));
					//TODO: restore: std::lock_guard<std::mutex> locker(m_mutex);  m_cached_data_size -= data_size;
				}
				catch (std::exception& e)
				{
					on_error(to_string() << "Failed to write frame. " << e.what());
				}
			});
			break;
		default:
			std::call_once(m_first_depth_flag, [&]()
			{
				try
				{
					const uint32_t device_index = 0;
					auto stream_type = frame_holder_ptr->frame->get_stream()->get_stream_type();
					auto stream_index = static_cast<uint32_t>(frame_holder_ptr->frame->get_stream()->get_stream_index());
					m_ros_writer->write_frame({ device_index, static_cast<uint32_t>(sensor_index), stream_type, stream_index }, capture_time, std::move(*frame_holder_ptr));
					//TODO: restore: std::lock_guard<std::mutex> locker(m_mutex);  m_cached_data_size -= data_size;
				}
				catch (std::exception& e)
				{
					on_error(to_string() << "Failed to write frame. " << e.what());
				}
			});
			break;
		}

    });
}
```

MartyG-RealSense, 2019-07-26:
If you are recording the very first frame, it is recommended that you allow the camera's auto-exposure to settle down first for the first several frames before beginning capture.

dorodnic, 2019-07-26:
Under `rs_export.hpp` there is a processing block that creates .bag from a single rs2::frameset, it might be useful if I understand the requirement correctly.

sailfish009, 2019-07-29:
@MartyG-RealSense , @dorodnic  i found that bag file itself is fine, but realsense viewer has bug.

1. i opened bag filed with "Load Recorded Sequence"
2. showing color image is white and depth image is normal.
3. close both image windows by click close icon.
4. toggle to start streaming data (depth and color) of "1-frame recorded bag file" 
5. re-opened image windows are just fine! (depth and color)

so i think my write_data_once function is fine.

sailfish009, 2019-07-30:
@dorodnic  i have tried save_single_frameset method, the result is like below:

color and depth image is abnormal.

```
std::thread video_processing_thread([&]() {
    while (alive)
    {
         // Fetch frames from the pipeline and send them for processing
         rs2::frameset data;
         if (pipe.poll_for_frames(&data))
         {
             if (b_save)
             {
                 b_save = false;
                 rs2::save_single_frameset single_saver; 
                 data.apply_filter(single_saver);
                 printf("single bag file saved!!!\n");
             }
         }
    // ...
    }
}
```

[image]

sailfish009, 2019-08-01:
these 1 frame bag file rendering solutions seems to be problematic, so i found and solved my request by using another method. so currently, i don't need this issue to be solved right now.

sailfish009, 2019-08-15:
this is not a solution, but work around:

in realsense viewer, when open 1 frame bag file, keep the specific device_model in viewer_model,
and make some bool variable flags(for run once) to close color stream channel.
then  in show_stream_header(), read the bool flag, and for color stream_type, dev->stop(viewer), and dev->play(...) again using the specific device_model from viewer_model.

basically, this just close color stream window, and play the stream channel again.

MartyG-RealSense, 2019-08-15:
Thanks so much for all the feedback you have provided about your workaround so that the community can benefit from the knowledge!

sailfish009, 2019-08-22:
found the reason, frameset size is fine. the cause is about frame timing. logically, write_data_once() function itself fine. but physically must have to wait sensor does it's job.

closed 2019-08-22 by sailfish009