realsenseai/librealsense #4512: unable to build librealsense2 for aarch64 nvidia jetson TX2 system

jawad-tech, 2019-07-26:
I am building librealsense2 library in Jetson TX2 but unable to build it. I am following the 

1) https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md 
2)  https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md

and try to run manually the ./patch-arch.sh and then cmake and make but it gives error related to hexfloat

" /home/nvidia/librealsense/librealsense/src/media/ros/ros_writer.cpp:296:52: error: ‘hexfloat’ is not a member of ‘std’
         frame_timestamp_msg.value = to_string() << std::hexfloat << std::fixed << pose->get_frame_timestamp();
"

Then I am using the installation tutorials as 
https://github.com/jetsonhacks/buildLibrealsense2TX

but on make the file in the build folder, like after 21 percent it gives error as :
" /home/nvidia/librealsense/librealsense/src/media/ros/ros_writer.cpp:296:52: error: ‘hexfloat’ is not a member of ‘std’
         frame_timestamp_msg.value = to_string() << std::hexfloat << std::fixed << pose->get_frame_timestamp();
"
Also I do not able to find sudo apt-get install librealsense2-dkms ????  but I found all libreasense2-dev or dbg libraries. 

I am following this thread https://github.com/IntelRealSense/librealsense/issues/2448 it also has the ssame error but i do not find the solution.

| Required Info                 
| Camera Model is D415
| Firmware Version                 
| Operating System & Version is UBUNTU 16.04 on Nvidia Jetson Tx2 with v4.4.38  
| Kernel Version (Linux Only) is 4.4.38
| Platform is Nvidia Jetson TX2 (aarch64 system)
  g++ version is 4.9.4

Note: I have successfully build the library on my x86_64 pc but have problem on aarch64 system.

dorodnic, 2019-07-26:
Hi @jawad-tech 

To get rid of compilation errors I think you need to update to GCC 5 as described here:
[installation.md#building-librealsense2-sdk](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk)

Kernel patching can be tricky since every flavor has its own patch. For most part you can skip this step while getting started. At the moment we only provide kernel patches for official LTS versions of Ubuntu. Rest are contributed and unfortunately we cannot update all patches on all platforms.
[jetsonhacks.com](https://www.jetsonhacks.com/) has good and mostly up to date scripts for getting started with RealSense on Jetson.

Also, 
Please try running the following commands:
```
sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE

sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u

sudo apt-get install librealsense2-utils
```
This is not *yet* official, but we are listening to community feedback and working on easy to install packages for the Jetson family (currently these are being tested, and at least for me they worked well on both TX2 and a Nano)

jawad-tech, 2019-07-26:
I have solved the compiler issue by upgrading to the gcc 5 yeah you are right 

Removing and purging the gcc 4.9.4 which was installed on my system
then follow the picture 
[image]

jawad-tech, 2019-07-26:
I have tried the the three steps you mentioned but there is still no librealsense2-dkms library

ev-mp, 2019-07-28:
@jawad-tech, the `librealsense2-dkms` package is available [strictly for X86 PCs](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#using-pre-build-packages).

Note that this package is required mainly if you're planning to utilize frame metadata attributes. In that case you have to build the SDK with `cmake ../ -DFORCE_LIBUVC=true`

jawad-tech, 2019-08-01:
hi, i have successfully installed it using your suggestions and updated the firmware but when the depth and stereo camera is functioning it says loss of packet frames, as i saw other thread it says the DKMS is required for complete fps but if I am using arm64 architecture then what should be the solution of incomplete frames received?

ev-mp, 2019-08-01:
@jawad-tech , the package loss is a long-traced phenomenon with Jetson-tx and it is not in the scope of librealsense2-dkms/ patches. The patches are required to allow some advanced features that the Realsense cameras can provide and they cannot affect the basic USB data flow that is handled by the board USB controller/chipset. These issues were previously reported on this forum as well, e.g. #2682.
I would refer you to the JetsonHacks forum for assistance.

The one thing to verify is checking the Realsense camera firmware and aligning it with the latest published/recommended version available on.

jawad-tech, 2019-08-15:
The issue is solved. The loss of packet frames is resolved by updating the firmware based on my other post. 
https://github.com/IntelRealSense/librealsense/issues/4539

closed 2019-08-15 by jawad-tech