realsenseai/librealsense #4513: Hardware_reset overflows /dev/media

whatisor, 2019-07-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } 
| Firmware Version                   | (5.11.1.100) 
| Operating System & Version |   Linux (Ubuntu16.04) 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.17.0 }                          | 
| Language                            |  {C++/SDK realsense-viewer itself }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Because of recommendation, I use hardware_reset to refresh device before connect.
However, each call of hardware_reset creates a new dev/media....
After multiple times(100-200 calls of hardware_reset), it will overflow Ubuntu  and return error in dmesg:
"could not get a free minor"
Then, my software and realsense-viewer never can query device(no device found).

dorodnic, 2019-07-26:
Hi @whatisor 
If I remember correctly, @ev-mp fixed this recently, the fix should be either in 2.24 or published in an upcoming release.

ev-mp, 2019-07-28:
@whatisor , this bug fix is [embedded into v2.24 release](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#bug-fixes).

closed 2019-07-29 by dorodnic

whatisor, 2019-08-01:
@dorodnic Thank for your quick reply.
How can I try it before release?
I guess I will need librealsense2-dkms new version.

ev-mp, 2019-08-01:
@whatisor , the fix was in the  `librealsense2` debian package.
In case you're still using the debian v2.17 version then  `sudo apt update && sudo apt upgrade` will fix it.

whatisor, 2019-08-02:
@ev-mp, it does not work.
/dev/media... still increases.
I am using Ubuntu 16.04 kernel 4.15.0.
librealsense2 2.15.0

I wonder why it is not librealsense2-dkms, because it is common problem when using video device such as USB Webcam.

ev-mp, 2019-08-04:
@whatisor, please double-check that you're using the correct librealsense version. In case that you're running ROS (you mentioned robotics) or any other SDK that is linked with librealsense then you need to make sure that the node you're using was built with the proper SDK version.

As for the `librealsense2-dkms`  - the issues was in the user space code. It is not related to kernel.

whatisor, 2019-08-05:
@ev-mp, I asked this because when I test unplug/plug with normal USB webcam, same issue occurs (/dev/media... keep increasing without release).
OR you mean it is different problem?

ev-mp, 2019-08-05:
If this happens in your application when you re-plug USB webcam, than this means that you have a resource management issue in code, especially  the device handles. All the video device file descriptors allocated in your code shall be released when the device is forcefully disconnected.

As a reference you can run `realsense-viewer` and activate `reset hw` continuously for 10-20 times while watching the devices list. The device are reused (`/dev/video0, /dev/video1 ...) and the count is not raised.

whatisor, 2019-08-05:
@ev-mp , I do same way for testing, not use my application.
What Ubuntu version are you using?
After I search around, it seems ubuntu 16.04 issue with media device control.

ev-mp, 2019-08-05:
@whatisor ,the tests are run on the the same PC as reported in the template : Ubuntu16 kernel 4.15.
Can you elaborate what do you mean by
> I do same way for testing, not use my application.

Realsense-viewer does not provide restart functionality for non-realsense devices. 
What application do you use to test the webcam? How is this related to librelasense?

whatisor, 2019-08-05:
@ev-mp, I want to say that issue is not only for realsense.
I do these test for 1 normal USB webcam (example logitec webcam), just **plug, unplug** many times.
Check /dev folder, I saw many /dev/media... files.

When you say you fixed it, I thought you modify usb driver( such as media controller,...) to fix this behavior.
In realsense, specifically realsense-viewer, call hardware_reset is same behavior with unplug and plug again.

whatisor, 2019-08-06:
FYI, I found report from Linux, but fixes is not released for Ubuntu 16.04
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v4.15.6&id=11474eb609f4a7469d85109347f74ee8972925fa