realsenseai/librealsense #4514: Intel RealSense Viewer error 0xc00d36b6

AntoineWefit, 2019-07-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | latest | 
| Operating System & Version |   Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  /                                     | 
| Platform                                 | PC  |
| SDK Version                            |  /                     | 
| Language                            |  /                       | 
| Segment			|  others                         | 

### The Viewer refused to stream with one of my cameras

> HResult 0xc00d36b6: "Cet objet doit �e initialis�vant que lop�tion demand�puisse seffectuer."

During some tests where I load different custom configurations, I was surprised to see the Viewer simply refusing to stream with one of my camera, because everything was working fine the minute before. I recorded the error but after restarting the Viewer, I couldn't reproduce it...

Don't really know what happened here, everything looks fine now.

[image]

RealSenseCustomerSupport, 2019-08-20:
----------------------------------------------

Hi Antoine,

Can you please provide us with the information about FW and Librealsense build version you have used and with .json files that you have used for custom presets as well.

Thanks you!

AntoineWefit, 2019-08-20:
@RealSenseCustomerSupport 
Firmware version is : 5.11.6.250
Librealsense build version is : 2.24
Config files : 
[28.txt](https://github.com/IntelRealSense/librealsense/files/3519081/28.txt)
[29.txt](https://github.com/IntelRealSense/librealsense/files/3519082/29.txt)

The thing is, I couldn't reproduce the error... Plus I don't know which problem is related to this "0xc00d36b6" error.
It happened once, kind of "out of nowhere", while I was testing some configurations.

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

Hi Antoine,

Thank you for the provided information!
We couldn't reproduce the issue. Please update us if you will experience it again.

closed 2019-09-12 by RealSenseCustomerSupport