realsenseai/librealsense #4515: Support of pyrealsense2 package on Python 3.7 (Windows)

sr2009, 2019-07-26:
Do we have support for pyrealsense2 package with Python 3.7? 

Currently I can load it in Python older version (e.g., 2.7) but not with 3.7.3 

I am in Windows 10 environment

MartyG-RealSense, 2019-07-26:
An update for Python 3.7 was added in SDK version 2.23.0.

https://github.com/IntelRealSense/librealsense/issues/4008#issuecomment-501565712

sr2009, 2019-07-30:
I do not see any support for PyPI package that will allow us to use "pip install pyrealsense2". The only option is to build from source?

MartyG-RealSense, 2019-07-30:
Information about installing the package can be found here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation

lramati, 2019-08-14:
We have found the issue in our CI and are working to address the problem.
We will notify when 3.7 compatible wheels become available on PyPi

ev-mp, 2019-09-26:
Python 3.7 wrapper was made available on Pypi with v2.28.0 release

closed 2019-09-26 by ev-mp

qaler, 2019-10-30:
Please update the readme doc in python wrapper, which indicates python3 is not supported by SDK and pypi.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

MartyG-RealSense, 2019-10-30:
Pip support for Python 3.7 was apparently added in the recent 2.28.0 SDK.

https://github.com/IntelRealSense/librealsense/issues/4765

cxbrooks, 2019-12-01:
Please update https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

and change 

> (Note that these binaries are built with Python 2.7, and cannot be import using Python 3).

to

> (Both Python 2.7 and Python 3 are supported).