realsenseai/librealsense #4516: No frame metadata available?

RohithKuditipudi, 2019-07-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   |  05.11.06.200| 
| Operating System & Version |   {Ubuntu 18.04}  | 
| Kernel Version (Linux Only)    |         4.15       | 
| Platform                                 | PC |
| SDK Version                            |  { 2.23 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {SLAM }                          | 

### Issue Description
Am trying to grab metadata (in particular sensor timestamp and frame number) from the frames I'm streaming from a D435i but it doesn't seem as though any metadata is available (i.e. rs2_supports_frame_metadata returns false for every metadata value). Apparently others have had this issue and the recommended fix was to set ENFORCE_METADATA to ON in the cmake file. However, it seems this option is only valid on Windows? Is there a way to enable metadata on Ubuntu? I'm currently trying this on my PC, but would also ultimately like to do something similar on a Nano or TX2.

ev-mp, 2019-07-28:
@RohithKuditipudi, the metadata enablement for Linux [requires kernel modifications](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#os-support), with either [manual kernel patching](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation) or installing [`librealsense2-dkms`](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution) package.

Currently the kernel patches are maintained for Ubuntu LTS versions for X86 PCs. For ARM-based boards you have to build the SDK with **libuvc** backend that provides similar functionality with no kernel changes

RohithKuditipudi, 2019-07-28:
@ev-mp Thanks for your response! Are there instructions for how to go about setting up the SDK with ARM-based boards? Because my end goal is to run this on a Nano (or TX2 if that would be easier). It seems like JetsonHacks does install something related to metadata, but then I have the same issue where rs2_get_frame_metadata always returns 0. So is there something additional I need to do?

ev-mp, 2019-07-29:
There are several guides for Arm boards in the [doc](https://github.com/IntelRealSense/librealsense/tree/master/doc) folder, including Raspberry, Odroid and [Jetson](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) boards.

Distribution-wise - recently we've started experimenting with Debian packages for Jetson, so you can install Librealsense v2.24 with `sudo apt install ...` ([link](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)). This is currently in pilot phase and does not include kernel patches, but other than that is a fully functional distribution. We'll consider for the next release to build it with `libuvc` backend so it will also support metadata

RohithKuditipudi, 2019-07-30:
@ev-mp From what I can tell it doesn't seem like the folder you linked to has instructions for how to enable metadata support? I followed the instructions on JetsonHacks for building librealsense on my Nano (including kernel patches), but still am not able to read metadata.

ev-mp, 2019-07-31:
@RohithKuditipudi , metadata support requires kernel patches, which are maintained by JetsonHack community as we maintain kernel patches for X86/X64 Debian distributions only. 
You should follow the [above instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#jetson-tx2-installation) to make the kernel changes for Nano/Jetson platforms.
On the other hand these community contributions are not regularly maintained and I'm not sure whether they up-to-date. Therefore I would definitely recommend to build and try the SDK first with the `libuvc` backend before modifying the kernel.

dorodnic, 2019-08-01:
I'm just gonna add a link to [libuvc_installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md). It is probably the easiest way to get metadata on Linux (there are some trade-offs, but these are relatively minor)

closed 2019-09-03 by dorodnic