realsenseai/librealsense #4517: "Frame didn't arrived within 15000"

skomatin, 2019-07-27:
####  Similar issue to #4071 except on macos

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | Latest | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Mac Book Pro 2015  |
| SDK Version                            |                             | 
| Language                            |  C++                          | 
| Segment			|                           | 

### Issue Description
According to realsense documentation, I should be able to "stream depth, infrared and color at all supported resolutions" on the D415 camera. I tested the following librealsense examples on xcode: rs-capture, hello-realsense and imshow (opencv wrapper) and all of them show the same error: "Frame didn't arrived within 15000" However when I disconnect and reconnect the camera the hello-realsense example works with a significant delay for a brief moment before this error occurs. 

#### Additional Info
I dual booted my computer with windows and tested all the above examples in visual studio and I had no issues and the sdk was working just fine. This goes to show that the hardware is capable of handling the frames. I also used it to update the camera's firmware to the latest version up the point of 1 week ago.

RealSenseCustomerSupport, 2019-09-17:
----------------------------------------------

@skomatin Sorry for the late response. What's the SDK and FW version did you use when you got the issue on MacOS? Any luck with the latest SDK and FW? Looking forward to your update. Thanks!

skomatin, 2019-09-17:
@RealSense-Customer-Engineering  Unfortunately I will not have access to the Realsense camera for another two weeks and hence I am not sure what the firmware version was. It was the latest firmware to the date of Aug 30 and no, I still experienced the same issue with that. 

As for the SDK I was unable to download the realsense-viewer SDK (if that's what you are talking about) on my macos. I used xcode IDE to test the camera as mentioned in my previous post.

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@skomatin Got you! Please provide SDK and FW version when you get back the camera. We'll also try to see if can reproduce here.

jursenbo, 2019-09-29:
I had the same problem with the SDK `v2.29.0` and FW `v5.11.15.00` on my `D435i` when running `hello-realsense` example. I could get the example working by changing the `pipeline` config to use `RS2_STREAM_DEPTH` stream type:
```
rs2::config config;
config.enable_stream(RS2_STREAM_DEPTH);
p.start(config);
```
Looks like another stream type is causing the timeout error.
A bit rough start with the `RealSense` camera, when even a `hello world` example does not run out of the box.

RealSenseCustomerSupport, 2019-10-23:
----------------------------------------------

@jursenbo ,

according to current release note,
https://github.com/IntelRealSense/librealsense/releases/tag/v2.29.0
D435i is supported on Linux, Windows 10 and Android. Mac OS support is scheduled next, and Windows 7 will be provided in future releases.

The issue in this ticket is a different one. Should double-check SDK/FW version or USB cable.

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@skomatin Any chance to try again? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@skomatin Do you have the camera now? Any update? Thanks!

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@skomatin Sorry that we didn't get response from you for weeks. Will close this issue accordingly. Please create new one if you still have questions/issues. Thanks!

closed 2019-11-22 by RealSenseCustomerSupport