realsenseai/librealsense #4519: realsense-viewer: error while loading shared libraries: librealsense2-gl.so.2.23 AFTER updating to 2.24

DCDCDCDCDCDC, 2019-07-28:
After install 2.24, the realsense-viewer still load librealsense2-gl.so.2.23.
i updated follow https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md.
and i don't know where am i wrong.
realsense-viewer: error while loading shared libraries: librealsense2-gl.so.2.23: cannot open shared object file: No such file or directory

ev-mp, 2019-07-28:
Hello @DCDCDCDCDCDC , are you installing Debians or building from source ?
In the former case this could be related to #4410, which should be resolved via `sudo apt update && sudo apt upgrade`,
When building from sources make sure to remove the previous version(s) with `sudo make uninstall` to avoid polluting the disk with multiple so files. In case the manifest files was accidentally removed (required by `sudo make uninstall`) -  you'll have to navigate to the `/usr/local/lib /usr/local/bin` directories and remove the leftovers manually.

If you have multiple librealsense versions installed intentionally then check that you invoke the proper `realsense-viewer` version. (`which realsense-viewer`) - I suspect that `usr/local/bin` path has precedence.

ev-mp, 2019-08-04:
@DCDCDCDCDCDC , did you follow the steps? 
Please update

DCDCDCDCDCDC, 2019-08-04:
@ev-mp, after totally reload the whole pkg, the problem solved. and ur right.

ev-mp, 2019-08-04:
Thanks for the confirmation

closed 2019-08-04 by ev-mp