realsenseai/librealsense #4521: How to set exposure and white balance value of RealSense from other RealSense which is set auto exposure and auto white balance mode .

wodndb, 2019-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Win10 | 
| Platform                                 | PC |
| SDK Version                            | 2.24.0 | 
| Language                            |  C#/Unity | 
| Segment			| others | 

### Issue Description
I use 3 RealSense cameras to scan object (inward facing). however, the colors of object in image from each camera are different. So I want to use same exposure and white balance value for all RealSense camera. But I want to use auto exposure and white balance mode. So I want to set exposure and white balance value of two RealSense from other RealSense which is set auto exposure and auto white balance mode. How can I do this?

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

Hi

It's very complicated to "sync" both AE and AWB. My suggestion would be:
1) Assuming your lighting source color temperature is fixed, you can calibrate one camera by using Manual White Balance (while AE enabled). And then align other cameras to same White Balance value. No more AWB mode for all cameras.

2) Same thing to AE unless all image scenes are exactly same. If they're different and you're trying to stitch images, it's better to use one camera as a reference too, and then use your app manually adjust other cameras' exposure to meet the reference exposure. This is supposed to be doable with RS API.

Thanks

wodndb, 2019-08-23:
@RealSenseCustomerSupport I appreciate your comments, I'm going to try what you recommended.

wodndb, 2019-09-05:
It's hard... I can't sync color of multiple realsense camera. I think it's good solution that add additional LED to normalize color of camera.

closed 2019-09-05 by wodndb