realsenseai/librealsense #4522: How can I change the distance of depth measurement?

rosesue, 2019-07-29:
As I remember, the distance of depth measurement can be changed for D400 serials Camera. I am wondering how I can change it, by realsense-viewer or through API call?
Another question is, is it possible to change the resolution (width, height, fps) in the runtime without stopping the streaming of video stream?
Thanks.

ev-mp, 2019-07-29:
@rosesue , you cannot change the distance, i.e depth measurements. You do have some control over how the depth is generated by using the [advanced mode API](https://intelrealsense.github.io/librealsense/doxygen/classrs400_1_1advanced__mode.html) . The  purpose of this API is fine-tuning  the camera to a specific environment (e.g outdoor, scanning, long range).
You can change the depth units/resolution with `RS2_OPTION_DEPTH_UNITS` option, is that what you mean? Then take a note that the depth units can be queried any time (e.g. `rs-enumerate-devices -o`) but modified only during streaming.

You cannot modify the resolution/fps on the fly.

ev-mp, 2019-08-04:
@rosesue, do you need more assistance on this ?

rosesue, 2019-08-07:
>  Then take a note that the depth units can be queried any time (e.g. rs-enumerate-devices -o) but modified only during streaming.
Did you mean I can change the depth unit on the fly? According to the spec of D435, the depth range is 0.11~10m. Does it limit the value of depth scale? In another word, I can only set depth unit from 0.00000168 to 0.000168?

RealSenseCustomerSupport, 2019-08-29:
----------------------------------------------

@rosesue, the RealSense D4 depth engine delivers 16bit depth data, with the default unit/scale of 1mm. The maximum depth unit is 10mm and can be changed on the fly.

Please keep in mind that "The depth error scales as the square of the distance away."

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-change-the-depth-step-size

Will close this ticket if there is no further question on this subject. Thank you

closed 2019-09-03 by dorodnic