realsenseai/librealsense #4524: [ANDROID] how 拖

CrazyHUAN, 2019-07-29:
hello ,I wonder  why class Sensor extends LrsClass instead of extends Options

I looked at the **rs.cpp** file .
`
struct rs2_sensor : public rs2_options
{
    rs2_sensor(rs2_device parent,
        librealsense::sensor_interface* sensor) :
        rs2_options((librealsense::options_interface*)sensor),
        parent(parent), sensor(sensor)
    {}
    ......
};
`
So how to set some parameters, such as setting the Option.EXPOSURE

Thank you

dorodnic, 2019-07-29:
Hi @CrazyHUAN 
The JNI bindings we have for Android are not one to one with C API and they are not yet complete (since we just introduced proper Android support couple of months ago)
So at the moment options API is missing AFAIK (@matkatz consider adding)
There should be another way to control this by loading custom JSON with specified control value. This I believe is exposed in JNI

CrazyHUAN, 2019-07-30:
Hi @dorodnic 
Thank you for your reply.

Yes, this can be done by loading custom JSON.
However the exposure values required in different lighting environments are different, so I need to adjust in real time. Therefore, I customize and expand Options, and no error is reported now, so I asked why.

This issue can be closed

closed 2019-07-31 by CrazyHUAN