realsenseai/librealsense #4525: Frame timestamp needs clarification

chutsu, 2019-07-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.10.13.0 | 
| Operating System & Version |  Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.15.0                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.24.0                          | 
| Language                            |  C++                          | 
| Segment			|  N/A                          | 

### Issue Description

[image]

This needs a little more claification. In particular I am interested in knowning for a **video frame**, this short sentence could mean either the start or the end of exposure time? Or is it just me?

ev-mp, 2019-07-29:
@chutsu, the `get_timestamp()` should be accompanied with `get_timestamp_domain` call.
In case the domain type is `HW Timestamp` then the former refers to the start of pixels readout ( USB transmit of data).
In order to get the start of exposure stamp you must use the metadata attributes.
See #2188 for details

chutsu, 2019-07-29:
> In case the domain type is HW Timestamp then the former refers to the start of pixels readout ( USB transmit of data).

Can we clarify this "start of pixel readout" corresponds directly to the start of exposure? It is not clear if there is a delay between the start of exposure and start of pixel readout. I don't know if the device has some form of buffer and thus incur a delay before the device decides to transmit the pixel data, which in that case means the `get_timestamp()` in the context means end of exposure.

ev-mp, 2019-07-29:
No. Readout begins after exposure is completed. See [the comment](https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803) in the referenced thread.
The hardware timestamps are all 32bits wide and are all zero-initiated. In order to convert the HW time to 64bit EPOCH you should probably use the global_time that was added since then. Note that in most cases you'll get this for free as this is the default Librealsense mode for `get_timestamp()` when the metadata is available.
The `Optical Timestamp` attribute of the metadata approximates the middle of the exposure phase in device clock. There is no additional timestamp for the beginning of exposuer phase.

Since the `Optical TS` and `HW TS` are stamped with the same clock, and also, the `HW TS` is automatically converted into EPOCH with the help of global_timestamp utility, then transitively you can calculate the `Optical TS` in EPOCH domain as well

chutsu, 2019-07-29:
I was unaware that `frame.get_timestamp()` and the metadata `RS2_META_DATA_FRAME_TIMESTAMP` were similar therefore I did not see the relevance in the referenced thread, my apologies. 

More comments are needed in the API docs for `frame::get_timestamp()`. Since the return seems to depend on the the time domain.

ev-mp, 2019-07-29:
Thanks for the feedback.
Two clarifications:  
- `frame.get_timestamp()` always converts the timestamps in milliseconds (floats), while the metadata attributes are always 32bit uint32 types.
- The domain provided by `frame::get_timestamp()` is partially controlled by user:
    - In case there is no metadata then the domain is automatically preset to "System (Epoch) Time".
    - If the metadata is enabled then the domain is either `HW Timestamp` or `Global Timestamp` based on the value of `RS2_OPTION_GLOBAL_TIME_ENABLED` option

chutsu, 2019-07-29:
^ Thats the kind of info I would love to see in the docs. Thanks for your help.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi Christopher,

Do you need any further clarifications regarding frame timestamp topic?

chutsu, 2019-08-23:
Not at this point. However the issue still persists, the API is not documented adequately to reflect different returns of `frame::get_timestamp()` on different time domains.

closed 2019-08-24 by RealSenseCustomerSupport

chutsu, 2019-08-24:
Issue isn't resolved and you closed it?

libdavid, 2019-10-03:
Hi @chutsu,

Did you have a chance to take a look at documentation changes @ev-mp have mentioned on Aug 27: 
https://github.com/IntelRealSense/librealsense/pull/4739? Please let us know whether it solves the issue.
Thank you!