realsenseai/librealsense #4526: Exporting and importing localisation map from realsense-viewer

ArvindSanjeev, 2019-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.0.18.5715) | 
| Operating System & Version |  Ubuntu 18.04.2  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.24 }                          | 
| Language                            |  {C}                          | 
| Segment			|  {others }                          | 

### Issue Description
Hello, I have been trying to export a localisation map from the realsense viewer. I let the T265 run for a few seconds while I try to move it around and capture the scene around it, I then stop the streaming in the realsense viewer, wait for 5-10 seconds and export the localisation map. It saves as a file without any extension.

I am able to import the file again, however when I start the streaming again in the realsense viewer, it doesn't seem to reposition itself in the earlier map. It just starts again from a new origin point as if the imported map was ignored. Can the realsense viewer import and reposition the T265 in the new map? If not, how can I do this?

P.S. I also tried adding a .raw extension to the saved map and tried the above steps with it as well, but no change in the result.

ev-mp, 2019-07-29:
@ArvindSanjeev , in the current implementation the device is not designed to "jump" to the locations found in the previous map, but will continue to run in its new "world". The old map can be used to query the location of the "static nodes"/ anchor points, though this use-case has not been elaborated yet.
@radfordi, can you comment?

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

@ArvindSanjeev , we are continuing updates to the relocalization feature and monitoring all the latest SDK changes and FW updates within the SDK would be the best option moving forward.

I've referenced a PR that also would be of assistance I think to you.
https://github.com/IntelRealSense/librealsense/pull/4321

giovannicairoli, 2019-09-09:
@RealSenseCustomerSupport Do you have any schedule about when it will be available? 2-6-12 months.

Thank you

RealSenseCustomerSupport, 2019-09-13:
----------------------------------------------

@giovannicairoli - We are not able to put a timetable on future changes at this time.  Please do keep an eye out on latest releases and release notes for updated features and bug fixes.

Thanks

closed 2019-09-26 by RealSenseCustomerSupport

peteroconnor-bc, 2020-01-22:
From my understanding, it does actually load the saved map but rotates/translates so that the camera still starts at the origin (correct?).  Is there any way to query the coordinate tranform between the previous saved map and the new one?  

@ArvindSanjeev said:

> The old map can be used to query the location of the "static nodes"/ anchor points,

How can we do this?

neilyoung, 2020-01-22:
> How can we do this?

In python:

`result, translation, rotation = pose_sensor.get_static_node("name")
`

whereas

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.pose)
device = config.resolve(pipeline).get_device()
pose_sensor = device.first_pose_sensor()
```

Similarly in C++

> Is there any way to query the coordinate tranform between the previous saved map and the new one?

There is not necessarily a need for having such a means. The return of `get_static_node` describes the transformation and rotation of your static node in the current camera reference frame.