realsenseai/librealsense #4529: Porting rs-pose example to nodejs

ijeppsen, 2019-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  0.0.18.5715 | 
| Operating System & Version |  Ubuntu LTS 18.04  | 
| Kernel Version (Linux Only)    |        4.18.0-25-                                  | 
| Platform                                 | PC|
| SDK Version                            |  latest (built from source)                         | 
| Language                            |  {nodejs }                          | 
| Segment			| VR/AR }                          | 

### Issue Description
I'm trying to port the rs-pose.c example to nodejs but having trouble getting the pose data. Currently the call   const pose_data = frameSet.poseData; returns undefined.  Any tips would be appreciated.

`'use strict';

const rs2 = require('../index.js');
const {GLFWWindow} = require('./glfw-window.js');
const {glfw} = require('./glfw-window.js');

const win = new GLFWWindow(1280, 720, 'Node.js Pose Example');

// Open a GLFW window
//const win = new GLFWWindow(1280, 720, 'Node.js PointCloud Example');
//const pc = new rs2.PointCloud();
const pipeline = new rs2.Pipeline();
const cfg = new rs2.Config();

cfg.enableStream( 'pose',-1, 0, 0, '6dof',0  );
 
pipeline.start(cfg);
console.log('Getting pose data.');
while (! win.shouldWindowClose()) {
  const frameSet = pipeline.waitForFrames();
  const pose_data = frameSet.poseData;
  console.log(pose_data)
}

pc.destroy();
pipeline.stop();
pipeline.destroy();
win.destroy();

rs2.cleanup();
`

MartyG-RealSense, 2019-07-29:
This topic was last discussed in April 2019.  Community member @eljefe40 apparently wrote his own version of rs-pose and was willing to post the code to the forum if requested.

https://github.com/IntelRealSense/librealsense/issues/3732

MartyG-RealSense, 2019-07-29:
This topic was last discussed in April 2019.  A community member called @eljefe40 was willing to post the code for his own version of rs-pose if asked.

https://github.com/IntelRealSense/librealsense/issues/3732

closed 2019-09-03 by dorodnic