realsenseai/librealsense #4532: T265 Hardware Synchronization between Camera and IMU

oonchulip97, 2019-07-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265                        | 
| Segment			|  Robotics, SLAM, ROS                        | 

### Issue Description
Is the fisheye images and IMU data from Intel RealSense Tracking Camera T265 hardware synchronized? This means that the timestamp of the raw images and the IMU are the same. This is because I would like to apply T265 onto third-party visual-inertial SLAM packages.

Razbotics, 2019-08-04:
Did you find answer? Is it hardware synchronised?

bfulkers-i, 2019-08-05:
The T265 accurately timestamps the sensor data (fisheye cameras and IMU) on the same clock. This clock is synchronized with the host to ~1ms accuracy. The rate of the fisheye cameras is 30Hz and the rate of the IMU data is 200Hz gyro, 62.5Hz accelerometer. The IMU part is a bosch BMI055. See also here for the coordinate system of each sensor: https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#calibration

shubhamwagh, 2019-08-07:
In rs_t265.launch file, there is a parameter called **enable_sync** which by default is currently set as  **false**. Change the value to **true** and check if the timestamps of the rawimages and IMU are same or not.

closed 2019-08-27 by dorodnic