realsenseai/librealsense #4534: Problem to install Firmware DFU on Ubuntu 18.04

SkyentificGit, 2019-07-31:
| Camera Model                       | D435| 
| Operating System & Version |   Ubuntu 18.04 | 
| Platform                                 | NVIDIA Jetson Nano  |

### Issue Description

To install the new firmware on D435 I followed the commands from: 
https://github.com/intelRealSense/librealsense/issues/2884

And I get the following problem:

$ sudo apt-get update
Hit:9 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                    
Hit:10 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease
Hit:11 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease
Hit:13 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
Hit:14 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Reading package lists... Done 
$ sudo apt-get install intel-realsense-dfu*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package intel-realsense-dfu*
E: Couldn't find any package by glob 'intel-realsense-dfu*'
E: Couldn't find any package by regex 'intel-realsense-dfu*'

Can you please help me?

k-maheshkumar, 2019-07-31:
Have tried [this](https://github.com/IntelRealSense/librealsense/issues/2884#issuecomment-472404067)? I just did and worked for me

s-hemer, 2019-07-31:
I guess the 'intel-realsense-dfu' package is only available for x86(-64) arch which the Jetson is not. In this case, I would suggest to use your desktop machine for rare firmware upgrades and not those specialized hardware boards.

s-hemer, 2019-07-31:
@k-maheshkumar He already referenced this topic in his question and also the apt output suggests that he already did this (bad) workaround.

SkyentificGit, 2019-07-31:
> Have you tried [this](https://github.com/IntelRealSense/librealsense/issues/2884#issuecomment-472404067)? I just did and worked for me

Yes, I tried. I also tried to do 
$ sudo apt-get install intel-realsense-dfu
without *. It does not work.

SkyentificGit, 2019-07-31:
> I guess the 'intel-realsense-dfu' package is only available for x86(-64) arch which the Jetson is not. In this case, I would suggest to use your desktop machine for rare firmware upgrades and not those specialized hardware boards.

This is the problem, I only have Mac and Windows 7 at work. And upgrade only works for Win 10 and Linux...

s-hemer, 2019-07-31:
Setup a lightweight VM or compile the DFU yourself on Jetson (as it seems to be part of librealsense now, cmp #3422).

s-hemer, 2019-07-31:
See [https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update](url), seems to be without bigger dependencies (tclap).

ev-mp, 2019-07-31:
@SkyentificGit , since Librealsense v2.24 the DFU (Firmware Update) routine [is fully integrated into the SDK](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements), available in  the viewer an also in a [command-line tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update) useful for scripting.

This new capability not only it makes the `intel-realsense-dfu` tool redundant, but also allows to update the firmware on any supported OS, including Android and MacOS. 
(Imho, in additional to the new functional capabilities it greatly improves the user experience, but  you can disregard this comment due to a conflict of interests :) )

SkyentificGit, 2019-07-31:
> See [https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update](url), seems to be without bigger dependencies (tclap).

This method worked!!! Thank you a lot!

SkyentificGit, 2019-07-31:
> @SkyentificGit , since Librealsense v2.24 the DFU (Firmware Update) routine [is fully integrated into the SDK](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements), available in the viewer an also in a [command-line tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update) useful for scripting.
> 
> This new capability not only it makes the `intel-realsense-dfu` tool redundant, but also allows to update the firmware on any supported OS, including Android and MacOS.
> (Imho, in addition to the new functional capabilities it greatly improves the user experience, but you can disregard this comment due to a conflict of interests :) )

This worked! Thank you! I think this method should be mentioned on the intel website. Otherwise it is almost impossible to find :)

dorodnic, 2019-08-01:
@SkyentificGit 

> I think this method should be mentioned on the intel website

You are right, we will. This method is new, so we didn't want to replace links to more mature tools before gaining more confidence in this solution.

SkyentificGit, 2019-08-01:
Thank you for the help!

closed 2019-08-01 by SkyentificGit