realsenseai/librealsense #4536: Point cloud modified when depth frame aligned

descamps2, 2019-07-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |   Ubuntu 18  | 
| Platform                                 | PC  |
| SDK Version                            |  2.23                         | 
| Language                            |  C++                       | 

### Issue Description

The point cloud calculated from the depth map, using the rs2::pointcloud class, is different when the depth frame is aligned to the color frame (using rs2::align). The points are shifted in x and y by a non constant value. z values remain the same. 
Is this expected behavior? What is the cause of this difference?

ev-mp, 2019-07-31:
@descamps2 , that's correct.
The rs2::pointcloud 's origin coordinate (x,y,z=0,0,0) is defined as the position of the left IR sensor, while the origin of the pointcloud produced from "Depth aligned to color" is the position of the RGB sensor.
Additionally , in the process of alignment the reprojected depth values are clamped to the RGB grid (aliasing), which adds up to the bias.

descamps2, 2019-08-01:
Ok, thanks for the clarification!

closed 2019-08-01 by descamps2