realsenseai/librealsense #4538: How to play back all the depth and color image frames without losing frames for post-processing?

nevynwong, 2019-08-01:
| Camera Model                       | { D400 } | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| Language                            |Matlab | 

When I use wait_for_frames to play back the bag file, .the frames will drop,so how can I get all the recorded frames for post-processing?

lramati, 2019-08-20:
Have you tried turning off real-time playback?

RealSenseCustomerSupport, 2019-09-16:
----------------------------------------------

@nevynwong Is this still issue for you? If yes, could you please reply to Iramati's question? Looking forward to your reply. Thanks!

nevynwong, 2019-09-16:
Sorry for the delay of email reply due to other things. This problem has been solved and can be closed.

closed 2019-09-17 by RealSenseCustomerSupport