realsenseai/librealsense #4539: Intel DFU not found

jawad-tech, 2019-08-01:
I am unable to solve the problem of installing DFU
I install in this way, however I cannot go ahead!!! My system is Ubuntu16.04 LTS and architecture is arm64. 

echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list 

sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE

sudo apt-get update

sudo pat-get install intel-realsense-dfu*

dfu command not found any answerss

ev-mp, 2019-08-01:
@jawad-tech , the debians are provided for X86 architecture only.
Note that the DFU is not necessary anymore as you have a much better alternative - #4534

jawad-tech, 2019-08-15:
I have solved this problem by:

1) First I disconnect my camera from arm64 computer and connect it to x86_64 architecture computer. 
2) Then follow the procedure of dfu install and firmware update as mention on Intel website pdf. 
3) Again connect the camera to arm64 pc and the problem is solved. 

Thanks.

closed 2019-08-15 by jawad-tech