realsenseai/librealsense #4540: UnityEditor's mouse drag freezes when I repeat play and stop in Unity

unagiHuman, 2019-08-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D400  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10| 
| Platform                                 | PC |
| SDK Version                            |    2.2.4                          | 
| Language                            |  unity                        | 
| Segment			|  others                         | 

### Issue Description
<bug report>
In Unity Wrapper project, if Unity is repeatedly played and stopped in Unity scene including RsDevice, UnityEditor will not be able to operate with mouse drag or Unity will crash.

How to reproduce

1. Install UnityWrapper
2. Open RealsenseSDK2.0 / Scenes / Samples / PointCloudProcessingBlocks
3. Connect Realsense D400 and repeat playback and stop of the scene of UnityEditor.
4. Operate with mouse drag. For example, dragging a GameObject on the Hierarchy, or attaching a GameObject to the Inspector.
5. The phenomenon that mouse dragging can not be performed or Unity crashes occurs.

wodndb, 2019-08-16:
Sometimes I also has same problem. Could you try to update Unity or RealSense SDK version?

unagiHuman, 2019-08-19:
I updated the librealsense version to 2.25, but a similar Unity freeze occurred. 
The version of Unity I tried when I reported above is 2019.1.13, but the phenomenon occurs regardless of the Unity version.

Nyro-Binag, 2019-08-29:
Same problem. The people at Nuitrack had the same problem with librealsence. https://community.nuitrack.com/t/unity-crashes-when-performing-a-drag-and-drop-operation/852/24

ohtaisao, 2019-08-30:
> 
> 
> Same problem. The people at Nuitrack had the same problem with librealsence. https://community.nuitrack.com/t/unity-crashes-when-performing-a-drag-and-drop-operation/852/24
I looked at the thread but it seems the same phenomenon. Moreover, NuiTrack seems to fix the bug by identifying the cause! I want Librealsense's UnityWrapper to fix it as well.

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi @unagiHuman,

The issue is currently under investigation.Updates will be posted as soon as investigation findings become available.

unagiHuman, 2019-12-24:
> 
> 
> Hi @unagiHuman,
> 
> The issue is currently under investigation.Updates will be posted as soon as investigation findings become available.

When will the issue raised in this issue be resolved? I hope it will be solved someday.

RealSenseCustomerSupport, 2020-01-08:
----------------------------------------------

Hi @unagiHuman,

Your request has been reported to our Program Office and they deal with this matter.We are sorry for the delay.

Thank you!

stents, 2020-03-27:
This is a major issue for us as well, any update on a fix for this? Does anyone have any work around? It is currently the main thing preventing us from using realsense cameras for any future projects, unless we can allocate time to debugging and fixing the issue ourselves.

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

Hi all,

The fix request is currently being processed. We apologize for the inconvenience this may have caused.

Thank you!

RealSenseSupport, 2020-10-30:
@unagiHuman @stents We didn't see this issue with the latest SDK v2.39. Could you please have a try? Looking forward to your update. Thanks!

RealSenseSupport, 2020-11-05:
@unagiHuman @stents Any update from your side? Thanks!

RealSenseSupport, 2020-11-16:
@unagiHuman @stents  Sorry that we didn't hear from you for weeks. Will close the issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-11-16 by RealSenseSupport

WarPromo, 2022-12-05:
fix pls