realsenseai/librealsense #4543: T265 MacOS 10.14.5 - Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFD (LIBUSB_ERROR_ACCESS)

RudyAramayo, 2019-08-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
T265/MacOSX/MacMini/latest master/objective-c/ Robot

When I open the rs-multicam sometimes it works, but sometimes I get this error:

22:28:12.361 [0] [E] Device-DC00: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFD (LIBUSB_ERROR_ACCESS)

I know that docs say MacOSX is not supported yet for T265. But can you please push the macOS T265 support, this wonderful camera is only sporadically working for me but the data is superb when I can get it. I just wish it was more reliable so I could actually use it from MacOS.

dorodnic, 2019-08-01:
@RudyAramayo 
Yes, understood. Core of the issue, I think, is not so much the T265 on Mac (should be working pretty well by now), but rather multi-camera on Mac. Are you using a single T265 (and no D400)? Did you try other tools?

RudyAramayo, 2019-08-01:
I only have a single t265 connected and no D400's I try all the rs-XXX tools to try to get any data at all and most give me the same response. Sometimes rs-enumerate-devices works, sometimes it doesn't.

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

@RudyAramayo - Are you using the latest LibRS?  Also wanted to make sure you're using USB3 connection along with a USB3 certified cable?

RudyAramayo, 2019-09-09:
Howdy, I am using a Mac mini 2012. The issue is that it’s an old computer. I am using he cable you guys provided and the usb 3 port on the Mac mini with no hub attached. A new vr ready windows machine connects with no issue at all. I wish the device would be as reliable on Mac as it is for windows because the droid I am working on is extremely visible to the world and would be a great showcase of your product. I need to update the computer eventually and I want the t265 chip to become a permanent robot addition to my heavy 100 lb droid.

Rudy Aramayo

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @RudyAramayo

I did see Mac Mini but didn't realize it was Mac mini 2012.  Sorry but we are not able to investigate on such an older model system.  Glad to hear though that the device is working on a newer machine without issue.

Is there anything else you need on this issue?

Thanks

closed 2019-10-11 by RealSenseCustomerSupport

RudyAramayo, 2019-10-24:
Howdy, I bought a brand new Mac mini 2019 and we still have the same issue. Its time to support Mac OSX 10.15 Catalina. This chip fails to be reliable but with rs-multicam I can see the data. It must be a threading issue that only happens when you do not open an OpenGL Window which seems to slow the race condition down so you can capture the USB device successfully. Just a guess.

2019-10-23 22:32:11.177036-0700 rs-multicam[11199:178209] Metal API Validation Enabled
2019-10-23 22:32:11.209139-0700 rs-multicam[11199:178616] flock failed to lock maps file: errno = 35
2019-10-23 22:32:11.209491-0700 rs-multicam[11199:178616] flock failed to lock maps file: errno = 35
22:32:16.535 [0] [E] Device-4600: MessageID 0x13 (DEV_STOP) failed with status 0x9
22:32:16.634 [0] [E] Device-4600: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
22:32:16.835 [0] [E] Device-4600: Entered state [ERROR_STATE]
Program ended with exit code: 0

I am even using the new development branch with the Xcode projects (thank you btw)

RudyAramayo, 2019-10-24:
rs-pose and others crash reliably... but rs-multicam gives me the 2 cameras.

RudyAramayo, 2019-10-24:
23:01:56.933 [0] [E] Device-1600: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
23:01:57.150 [0] [E] TrackingManager: Stopping Manager - Cleaned 1 non complete callbacks (onVideoFrame, onPoseFrame, onGyroFrame, onAccelerometerFrame, onControllerframe) - latency is too high!
23:01:57.150 [0] [E] Device-1600: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
23:01:57.150 [0] [E] Device-1600: Entered state [ERROR_STATE]

RudyAramayo, 2019-10-24:
23:04:57.419 [0] [E] Device-2000: ERROR: Bulk transfer message 0x25 (DEV_SET_LOW_POWER_MODE) response to host got LIBUSB_ERROR_IO. Host did not return answer
23:04:57.420 [0] [E] Device-2000: ERROR LIBUSB_ERROR_NO_DEVICE while control transfer of messageID: 0x11 (CONTROL_GET_INTERFACE_VERSION)
23:04:57.420 [0] [E] Device-2000: Error Transferring CONTROL_GET_INTERFACE_VERSION
23:04:57.420 [0] [E] Device-2000: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_OTHER. Bytes requested 6, bytes transferred 0
23:04:57.420 [0] [E] Device-2000: USB Error (0x6)
23:04:57.420 [0] [E] Device-2000: Error: Failed to get device info (0x6)
23:04:57.420 [0] [E] Device-2000: ERROR LIBUSB_ERROR_NO_DEVICE while control transfer of messageID: 0x11 (CONTROL_GET_INTERFACE_VERSION)
23:04:57.420 [0] [E] Device-2000: Error Transferring CONTROL_GET_INTERFACE_VERSION
23:04:57.420 [0] [E] Device-2000: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_OTHER. Bytes requested 6, bytes transferred 0
23:04:57.420 [0] [E] Device-2000: USB Error (0x6)
23:04:57.420 [0] [E] Device-2000: Error: Get Device Info error (0x6)
23:04:57.420 [0] [E] Device-2000: Entered state [ERROR_STATE]
23:04:58.957 [0] [E] Device-5000: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
23:04:59.170 [0] [E] TrackingManager: Stopping Manager - Cleaned 1 non complete callbacks (onVideoFrame, onPoseFrame, onGyroFrame, onAccelerometerFrame, onControllerframe) - latency is too high!
23:04:59.170 [0] [E] Device-5000: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
23:04:59.170 [0] [E] Device-5000: Entered state [ERROR_STATE]
Program ended with exit code: 9

radfordi, 2019-11-08:
If you've had problems with T265 on a Mac previously, then #5213 will likely address them.