realsenseai/librealsense #4547: firmware update error of D435

seunghak-kim, 2019-08-01:
camara model : D435
firmware version : 05.11.06.250
OS : ubuntu 18.04 LTS

I want to downgrade the firmware to 5.10.03. However, I installed DFU and followed the instructions, but I get the following error:
---------------------------------------------------------------------------------------------------------------------------------
wireless@ubuntu:~/Downloads$ intel-realsense-dfu -b 001 -d 018 -f -i /Downloads/Signed_Image_UVC_5_10_3_.bin

 - Intel® RealSense™ DFU Tool v1.1.0-

Device busnum = 1
Device devnum = 18
FW file path = /Downloads/Signed_Image_UVC_5_10_3_.bin

Error: failed to open FW file, error = -2
---------------------------------------------------------------------------------------------------------------------------------

Please help me solve it.

dorodnic, 2019-08-01:
/Downloads doesn't look right, maybe ~/Downloads or ./Downloads or /user/yourusername/Downloads? 
Also, you can try just doing it from the latest Viewer, option to burn specific firmware is under sensor "hamburger" icon (the three stacked lines)

seunghak-kim, 2019-08-04:
thank you!!!
I solved it.

closed 2019-08-06 by dorodnic