realsenseai/librealsense #4550: [Question] Tune the D435 for more depth accuracy at very close range ( < 20cm ~ 40cm)

gachiemchiep, 2019-08-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 and D435i  | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-55-generic                                     | 
| Platform                                 | PC  |
| SDK Version                            | 2.24 }                          | 
| Language                            |  Cpp / Python                           | 
| Segment			|  Robot                          | 

### Issue Description

Hello Realsense team. 
I'm trying to build a bin-picking robot with D435 as main camera and Universal Robot as robot's arm. 

This is not an impossible task, i found 2 working demo on youtube. 
1. [Using D435 demo](https://www.youtube.com/watch?v=NdEA-esYW9I) 
2. [Using rc visard 65](https://www.youtube.com/watch?v=-IeZbn1aA8Q)
Please be notified that, the "rc visard 65" sensor have a depth measurement range in "0,2 m – 1,0 m". The "D435" sensor have a depth measurement range in "0,1m ~ few m ". So theoritically, D435 could give better depth image than "rc visard 65". 

I try the [Best-Known-Methods for Tuning Intel® RealSense™ D400 Depth Cameras for
Best Performance](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) and [Intel preset](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) but the result is not as good as  rc visard 65.  Please watch the  [Using rc visard 65](https://www.youtube.com/watch?v=-IeZbn1aA8Q) from 37s to 50s, the depth image appear noseless. But in my case, the result is as follow : 

[image]

My question is "how to tune the D435 for more depth accuracy at very close range ( < 20cm ~ 40cm)" ? 
As in https://github.com/IntelRealSense/librealsense/issues/3569 , the D435 has factory-calibration parameters. Do i need to do the calibration parameters in this case too ? 

Thank you in advance

agrunnet, 2019-08-23:
It should look a lot better. I will send some example. 
Here is 848x480 @ 30fps. Laser power reduced to 70mW to remove saturation. 
[image]

agrunnet, 2019-08-23:
Here is point cloud with post processing, with 2x down-sample, and some temporal and spatial averaging. 
[image]

agrunnet, 2019-08-23:
I was using AlphaX=0.7 and deltaX=60 and AlphaT=0.2 and deltaT=20 when post-processing. 

It looks like your images may be saturated by laser so turn it down. 
Also the minimum distance at 848x480 is ~18cm so dont move closer than that. In one part of image you on the right you may be closer. 
If there are specific scenes and features you need us to try then show us a picture and we can try to reproduce.

agrunnet, 2019-08-23:
Example video, just for reference.  I know video quality is not great. 

[D435_848x480.zip](https://github.com/IntelRealSense/librealsense/files/3535866/D435_848x480.zip)

gachiemchiep, 2019-08-25:
Hello @agrunnet 
THank you for your answer. 

I notify that the color ruler in your picture is different from mine.
1. mine : 0 ~ 4
2. yours: 0 ~ 1

Maybe this is the reason for this difference. I'm using realsense-viewer. Is there any options that i should change to make my color rules is as same as yours ?

```bash
I was using AlphaX=0.7 and deltaX=60 and AlphaT=0.2 and deltaT=20 when post-processing.
```

I'm totally lost at this point. What kind of filter options are your mentioning about ?
1. AlphaX=0.7 and deltaX=60 
    -> ???
2.  AlphaT=0.2 and deltaT=20 
   -> i guess this is the Temporal filter

agrunnet, 2019-08-26:
Color setting should not matter for depth. 
The other depth setting are the “post processing” as outlined in the post processing white paper. You should be able to find the post processing in the RealSense viewer. I am sorry if I confuse with naming but it should be roughly the same in the viewer.

gachiemchiep, 2019-08-28:
Hello @agrunnet 
Could you share your preset file ?

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi,
There is no preset file.
Please use

1. AlphaX=0.7 and deltaX=60
-> Teporal Filter
2. AlphaT=0.2 and deltaT=20
-> Spatial Filter

RealSenseCustomerSupport, 2019-10-03:
----------------------------------------------

Hi,
Do you need further support on this topic?

gachiemchiep, 2019-10-04:
hello @RealSense-Customer-Engineering 
Thank you for your support . I'm good to go.

closed 2019-10-04 by gachiemchiep