realsenseai/librealsense #4551: Timestamp problem when using wait for frames

yangliu995, 2019-08-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   |  | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.23.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<I want to capture consecutive frames using pipeline.wait_for_frames() and depth_frame = frames.get_depth_frame(), and I recorded the timestamp of each obtained depth_frame. The problem is some of the neighboring frames have same timestamp, why would this happen? Following are my code and plot of timestamp:

import pyrealsense2 as rs
import numpy as np

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

#Declare pointcloud object
pc = rs.pointcloud()
#Declare point object
points = rs.points()

# Start streaming
profile = pipeline.start(config)
device = profile.get_device()
depth_sensor = device.first_depth_sensor()

timestamp = []
deltaTime = [0]

for j in range(100):
    # Wait for a set of frames
    frames = pipeline.wait_for_frames()
    # Retrieve the first depth frame, if no frame is found, return an empty frame instance.
    depth_frame = frames.get_depth_frame()
    # record the timestamp of retrieved frame 
    timestamp.append(depth_frame.timestamp)
    # delete variable
    del depth_frame,frames

pipeline.stop()

[image]

ev-mp, 2019-08-05:
@yangliu995 , the pipe object of the SDK is designed to provide an aggregation container (frameset) for all the enabled stream (Depth+RGB). And when only one of the streams (e.g. RGB) is updated on time then the generated frameset will include the new RGB and the previous Depth frames (i.e. repetitions).
In case you comment out the RGB config line then the duplications shall disappear from the resulted graph.
>#config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

yangliu995, 2019-08-06:
@ev-mp Thanks a lot for your reply, that's very helpful. So that means if I enable both RGB and Depth streams, sometimes the depth stream can not be updated on time? If I want to keep RGB stream, is there any way to make both streams update synchronously?

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

Hi @yangliu995,

Since the hardware sync between depth and color is applied for D415 only, I think you need to manipulate the frame data by your own thru frame number or timestamp.

ev-mp, 2019-09-19:
@yangliu995 , streaming RGB+Depth synchronized can be achieved when you follow certain rules:
- RGB and Depth should be set to identical FPS.(15/30)
- The Exposure priority control shall be set to prefer FPS over AE  (and prevent auto-exposure from affecting the FPS). Or in case of the manual exposure - the value must be lower than the expected frame interval.

Adhering to those rules would allow D435i to activate and effectively employ the internal RGB->Depth sync with the RGB sensor acting as Master. It is still possible though, for the firmware to temporally deactivate the internal sync due to additional constrains, but in most cases you'll get HW-synchronized depth and RGB frame pairs.

ev-mp, 2019-09-26:
@yangliu995 , do you need it further clarified ? Please update

yangliu995, 2019-10-14:
Thanks a lot. This helped me solve my problem.

closed 2019-10-14 by yangliu995