realsenseai/librealsense #4552: Why did our D435 camera sometimes lose contact?

Dontla, 2019-08-02:
We use a win10 system computer to process the transmission data of six D435 cameras at the same time. Because the usb3.1 cable of the camera is not long enough, we use six USB3.0 cables to extend, but we find that in the process of using, the camera will lose contact, as can be seen in the device manager can not find the lost connection. D435 camera, unless we re-plug the USB interface, it can be restored to normal, we do not know what caused it.

I don't know how to describe it accurately, so I drew a picture about how the camera connects to our computer:

[image]

As I described in my last email, our default USB cable is not long enough, so we use one extension cable (In my picture I signed extend cable) for each camera. Its model is USB 3.0, through which we can connect to our computer. And six cameras are linked in exactly the same way.

This picture is the interface picture when we connect the USB extension cables to our computer:
[image]

 - The following pictures is our computer configuration:
 
[image]
[image]

(We don't know why one of the D435 cameras is read as 430, nor is it possible to reinstall the firmware using official DFU tools. We don't know if it will affect our use.)

[image]
[image]

 - The following picture is the display information of our cameras in the Intel Realsense Viewer:
 
[image]
 [image]

 We don't know why USB cables are identified as 3.2, and sometimes cameras went wrong:
 [image]
[image]

Once the camera can't be connected, our program will not run, nor can we restart the program. Unless we pull out and re-plug the USB cable interfaces, the problem can be solved, but the time when the camera loses its connection is uncertain, and we can not interfere with the operation of the device manually in actual use, so this is really a troublesome problem for us.

And sometimes, we also occur these problem:

like: camera scp overflow 
like: camera has connected and disconnected and connected and disconnected...all the way...

MartyG-RealSense, 2019-08-02:
Because RealSense cameras transmit a high volume of data through the USB cable, extending the cable beyond 2 m can result in a loss of performance or loss of the camera connection altogether unless you have high quality cabling.

Newnex supply industrial-grade USB cables and equipment such as extenders and fiber optics that can extend the camera up to 100 m.  They are not inexpensive though.

[https://www.youtube.com/watch?v=GLQgR1jT04M](url)

Alternatively, a member of the RealSense team at Intel developed a system called EtherSense for using ethernet to allow multiple cameras to be connected over a distance greater than 30 m.

[https://github.com/krejov100/EtherSense](url)

Ethernet has also been used by the CONIX Research Center at Carnegie Mellon for doing point cloud stitching with large numbers of RealSense cameras.

[https://github.com/conix-center/pointcloud_stitching](url)

In regard to the difficulty of doing unplug-replug of USB, a **hardware_reset** instruction can be incorporated into RealSense scripts to help avoid the need for this.  If your project relies on the RealSense Viewer though, it would be hard to make use of this unless you edit the Viewer's source code to include the instruction and then compile your own custom version of the Viewer from that code.

pwalczykk, 2019-08-05:
Hey, 

I believe that we might have a related problem with 3-4 Realsense D415 cameras connected to a single PC running on Linux. From time to time, one or more realsense camera is not detected by **rs-enumerate-devices** or causes **rs-enumerate-devices** to return with an error. Unplugging and replugging the camera usually solves the issue. 

What we have already tested:
- using a PCI-E card with 4 separate USB 3.0 controllers
- using industrial-grade, 4.6m USB 3.0 cables from TetherTools (and few others)
- using a fiber-optic link connection

It seems that any of those solves our problem, but using high-grade equipment reduces the frequency of the issues. 

I have the following questions:
1. What is **hardware_reset** actually doing? Can it reset the camera if it was not detected by **rs-enumerate-devices**? Is there any other way to reset the camera without unplugging it or cutting out its power supply?
2. I have managed to run Realsense D415 with Nvidia Jetson Nano and stream its RGB and D channels through Ethernet using ROS. Are you aware of any potential issues that might arise while using a network of single-board computers with one Realsense camera attached to each of them?

MartyG-RealSense, 2019-08-05:
1.  hardware_reset acts the same as using the hardware reset option in the RealSense Viewer or unplugging-replugging the USB cable.  So I would expect non-detection by rs-enumerate-devices not to be an obstacle to a successful reset, since the point of a reset is to get it to detect the camera when it wasn't detectable before.

Yes, I would definitely recommend high-grade / industrial-grade equipment as a means of reducing the likelihood of problems occurring.  for example, industrial-grade USB cables may be used on mission-critical equipment such as hospital medical scanners, where even one error would not be acceptable.

If you are using Linux then you may be able to write a program to reset a particular USB port.  Here's an example:

https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

2.  Your setup sounds very similar to that of the EtherSense ethernet system that a RealSense employee at Intel developed.  

https://github.com/krejov100/EtherSense

Although your setup has a good chance of working if it follows along the lines of that project, there is always the potential for issues to be introduced through the particular hardware that you are using, of course.  Assuming that the singe-board computer works fine, I would say that you should:

-  Make sure that the cameras are provided with enough power on the USB ports to ensure stable running.  Since the Jetson Nano has full USB 3 'A' ports, you are not likely to encounter problems with that though.

-  If your setup uses camera sync cables and those cables are long, include components in the cable that protect against electrostatic discharge (ESD).  Otherwise, the D415 may be vulnerable to its counters resetting and the RGB freezing up if an ESD discharge occurs.

pwalczykk, 2019-08-05:
1. There is one thing I don't really understand. How can I send a hardware_reset signal to a camera that is not being detected? Is there a way to do it with Realsens-Viewer?

I know that we were resetting USB ports with some different methods earlier and the overall feeling was that it does not solve the issue. Anyway, I have reproduced a problem with only 3 cameras detected out of 4 connected. Then I have used one of method found under your link. The missing camera was detected after resetting USB controller. Thanks! 
```
Ada:~$ rs-enumerate-devices | grep Serial
    Serial Number                 : 	802212060668
    Serial Number                 : 	802212061216
    Serial Number                 : 	811112060576
Ada:~$ sudo su
root:~$ for i in /sys/bus/pci/drivers/[uoex]hci_hcd/*:*; do>   [ -e "$i" ] || continue
>   echo "${i##*/}" > "${i%/*}/unbind"
>   echo "${i##*/}" > "${i%/*}/bind"
> done
root:~$ exit
Ada:~$ rs-enumerate-devices | grep Serial
    Serial Number                 : 	811112060835
    Serial Number                 : 	802212060668
    Serial Number                 : 	802212061216
    Serial Number                 : 	811112060576
```
I will make more experiments to check if this method repeatably solves all our problems, this will take some time. I will come back with some results probably in a few days.

2. Thanks for the description, I assume there would be no major blockers as we are not using sync option. But if resetting USB controllers solve our problem, there will be no point in adding more hardware.

MartyG-RealSense, 2019-08-05:
The hardware_reset instruction can be put near the start of a script to reset the camera if it loses connection and it is not practical to use methods such as a Viewer reset or unplug-replug.  for example, in some industrial cases the computer will be locked in a cabinet where employees cannot access the USB ports if a problem occurs, so it has to be resettable automatically.

It is simple to do a hardware reset in the RealSense Viewer.  Left-click on the 'More' icon at the top of the Viewer's options side-panel and select the 'Hardware Reset' option from the menu.

[image]

I've had a couple of other cases recently where one out of a batch of cameras (such as 1 of 4 or 1 of 5) were not being detected.  A method that restored detection for those users was to recalibrate them with the Dynamic Calibrator tool.  Page 14 of its user guide explains how to set it up for Linux.

https://www.intel.co.uk/content/www/uk/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

pwalczykk, 2019-08-05:
> The hardware_reset instruction can be put near the start of a script to reset the camera if it loses connection and it is not practical to use methods such as a Viewer reset or unplug-replug. For [...]

Sure, I was just looking for a quick way to test it. 

> It is simple to do a hardware reset in the RealSense Viewer. Left-click on the 'More' icon at the top of the Viewer's options side-panel and select the 'Hardware Reset' option from the menu.

This option is only available if realsense camera is being detected. In my case, the camera is not showing up in realsense-viewer and I can not add it "Add Source" button. Hence I do not have access to "More/Hardware Reset".

Comming back to the USB reset...
I have managed to break realsense camera 5 times in a row, in a way, that restarting USB xhci drivers does not help (only unplugging works). I changed cameras and cables that broke between these tests.

It was as simple as running a script that reset USB drivers, while all 4 camears were streaming RGB and D to RealsenseViewer. Here is a quite verbose output of a script (available cameras are listed with `rs-enumerate-devices | grep Serial`):
```
Available Realsense cameras:
    Serial Number                 : 	811112060835
    Serial Number                 : 	802212060668
    Serial Number                 : 	802212061216
    Serial Number                 : 	811112060576
Unbinding...
echo 0000:19:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1a:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1b:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1c:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
Binding...
echo 0000:19:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1a:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1b:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1c:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
USB restart finished. Available Realsense cameras:
    Serial Number                 : 	811112060835
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
```
However, if the cameras are not streaming, and were not streaming during the restart, reseting USB xhci seems to work. Especially if cameras were not showing up after replugging. In that case, I got at least 5 successful usages of unbinding and binding driver. Example:
```
Available Realsense cameras:
    Serial Number                 : 	746112060395
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
Unbinding...
echo 0000:19:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1a:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1b:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1c:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
Binding...
echo 0000:19:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1a:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1b:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1c:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
USB restart finished. Available Realsense cameras:
    Serial Number                 : 	811112060835
    Serial Number                 : 	746112060395
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
```

My first conclusions would be following: to properly reset Realsense camera with xhci_hcd/unbind, it should not be in the image streaming state.

MartyG-RealSense, 2019-08-05:
Thanks so much for your very detailed sharing of what worked for you!

pwalczykk, 2019-08-06:
No problem, I would really like to make those cameras work reliably.

During the night tests, we have detected one example of a camera that hung in a way that it was impossible to bring it back with USB drivers reset. After killing a docker container with 3 realsense-ros nodes that were handling 3 cameras, one of them "disappeared". After starting a new container, it could only see two cameras. I wonder now if the wrong termination of the realsense-ros node can cause a hanging camera issue.  

Also, I would like to gather more debugging information from realsense cameras during my tests. I know that rs-fw-logger could provide some binary information, but I am unable to read them. Do you have any suggestions?

MartyG-RealSense, 2019-08-06:
Thanks for the additional insights from your tests.   :)

Intel has internal tools for decoding log files submitted to them.   They aren't meant to be read by the user.

https://github.com/IntelRealSense/librealsense/issues/1215#issuecomment-367922389

pwalczykk, 2019-08-06:
If I record some rs-fw-loggers logs from the moment when a camera broke, where can I post them to be analyzed?

MartyG-RealSense, 2019-08-06:
Usually, log files are provided to Intel by attaching them to a comment.  You can do this by drag and dropping the file onto the comment writing window.

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

Hi

Is there any chance to upload the log file? or do you want us to close this ticket?

Thanks

pwalczykk, 2019-09-06:
Hi, 

sorry for not being verbose enough. I have shared a code and hardware setup of an experiment that is capable of reproducing hanging camera problem. Link to download sample log file is in the experiment description. Everything should be described in issue #4750, referenced just above. 

"The experiment gathers logs from roslaunch and rs-fw-logger, which might be useful for debugging purpose. A sample of logs from 64-hour experiment, which contains 45 examples of hanging camera issue is available under [this link](https://drive.google.com/file/d/1oQOymr_f1ylLZ8w_Q1kWSmGma9xG92Q9/view?usp=sharing)"

pwalczykk, 2019-09-06:
I am attaching a log from from first camera failure detected in mentioned 64-hour experiment:

Waiting for RealSense device to connect...
RealSense device was connected...
Device Name: Intel RealSense D415
Device Location: /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:04.0/0000:1c:00.0/usb10/10-1/10-1:1.0/video4linux/video12

2019-08-24 00:15:19  FW_Log_Data:0F 00 00 00 A0 01 0A 10 8B 02 72 22 2B 00 28 00 00 00 00 00 77 B4 9B D7 A0 A1 1D 10 97 00 44 34 03 00 01 00 10 00 00 00 98 05 D7 DE A0 01 0D 10 5B 00 1F 42 0B 00 0F 00 00 00 00 00 F1 10 D7 DE A0 01 0D 10 10 01 24 52 00 00 00 00 00 00 00 00 C7 13 D7 DE A0 A1 1D 10 97 00 44 64 03 00 01 00 10 00 00 00 8C B6 EA DE A0 A1 1D 10 97 00 44 74 03 00 01 00 30 00 00 00 6C 0B FF DE A0 01 0D 10 5B 00 1F 82 02 00 10 00 00 00 00 00 E9 15 FF DE A0 01 0D 10 30 01 24 92 00 00 00 00 00 00 00 00 BF 18 FF DE A0 A1 1D 10 97 00 44 A4 03 00 01 00 10 00 00 00 AC 5D 13 DF A0 01 0D 10 10 01 24 B2 00 00 00 00 00 00 00 00 19 69 13 DF 

2019-08-24 00:15:19  FW_Log_Data:0F 00 00 00 A0 01 0D 10 0F 01 24 C2 00 00 F4 01 00 00 00 00 94 41 31 E1 

2019-08-24 00:15:22  FW_Log_Data:0F 00 00 00 A0 01 0A 10 8B 02 72 D2 2B 00 28 00 00 00 00 00 4A 3C 27 13 

2019-08-24 00:15:25  FW_Log_Data:0F 00 00 00 A0 01 0A 10 8B 02 72 E2 2A 00 27 00 00 00 00 00 04 C4 B2 4E 

2019-08-24 00:15:28  FW_Log_Data:0F 00 00 00 A0 01 0A 10 8B 02 72 F2 29 00 27 00 00 00 00 00 B6 4B 3E 8A 

2019-08-24 00:15:28  FW_Log_Data:0F 00 00 00 A0 A1 1D 10 97 00 44 04 03 00 01 00 10 00 00 00 DC 26 EA 91 A0 01 0D 10 5B 00 1F 12 56 00 0F 00 00 00 00 00 35 32 EA 91 A0 01 0D 10 10 01 24 22 00 00 00 00 00 00 00 00 0B 35 EA 91 A0 A1 1D 10 97 00 44 34 03 00 01 00 10 00 00 00 B8 7D F7 91 A0 A1 1D 10 97 00 44 44 03 00 01 00 30 00 00 00 E0 2F 0B 92 A0 01 0D 10 5B 00 1F 52 02 00 10 00 00 00 00 00 5D 3A 0B 92 A0 01 0D 10 30 01 24 62 00 00 00 00 00 00 00 00 33 3D 0B 92 A0 A1 1D 10 97 00 44 74 03 00 01 00 10 00 00 00 70 C7 20 92 A0 01 0D 10 10 01 24 82 00 00 00 00 00 00 00 00 DD D2 20 92 

2019-08-24 00:15:28  FW_Log_Data:0F 00 00 00 A0 01 0D 10 0F 01 24 92 00 00 F4 01 00 00 00 00 7B 7A AE 92 

2019-08-24 00:15:28  FW_Log_Data:0F 00 00 00 A0 A1 1D 10 97 00 44 A4 03 00 01 00 10 00 00 00 7C 7F 40 95 A0 01 0D 10 5B 00 1F B2 02 00 0F 00 00 00 00 00 D5 8A 40 95 A0 01 0D 10 10 01 24 C2 00 00 00 00 00 00 00 00 AB 8D 40 95 A0 A1 1D 10 97 00 44 D4 03 00 01 00 10 00 00 00 A4 D7 4D 95 A0 A1 1D 10 97 00 44 E4 03 00 01 00 30 00 00 00 44 87 61 95 A0 01 0D 10 5B 00 1F F2 02 00 10 00 00 00 00 00 C1 91 61 95 A0 01 0D 10 30 01 24 02 00 00 00 00 00 00 00 00 97 94 61 95 A0 A1 1D 10 97 00 44 14 03 00 01 00 10 00 00 00 7C 7D 76 95 A0 01 0D 10 10 01 24 22 00 00 00 00 00 00 00 00 E9 88 76 95 

2019-08-24 00:15:28  FW_Log_Data:0F 00 00 00 A0 01 0D 10 0F 01 24 32 00 00 F4 01 00 00 00 00 57 5D D1 96 

2019-08-24 00:15:28  FW_Log_Data:0F 00 00 00 A0 A1 1D 10 97 00 44 44 03 00 01 00 15 00 00 00 38 60 C8 99 A0 01 0D 10 5B 00 1F 52 02 00 0F 00 00 00 00 00 22 6D C8 99 A0 01 0D 10 15 01 24 62 00 00 19 00 00 00 00 00 F8 6F C8 99 A0 A1 1D 10 97 00 44 74 03 00 01 00 15 00 00 00 54 B8 D5 99 A0 A1 1D 10 97 00 44 84 03 00 01 00 2C 00 00 00 04 68 E9 99 A0 01 0D 10 5B 00 1F 92 02 00 15 00 00 00 00 00 98 73 E9 99 A0 01 0D 10 2C 01 24 A2 00 00 09 00 00 00 00 00 6E 76 E9 99 A0 A1 15 10 90 00 BE B2 04 00 04 00 5A 00 00 00 D0 6C 0F 9A A0 A1 15 10 8B 00 18 C1 01 00 01 00 5A 00 00 00 FD 54 11 9A A0 A1 15 10 8C 00 18 D1 06 00 07 00 5A 00 00 00 83 DF 13 9A A0 A1 15 10 90 00 BE E2 04 00 04 00 5A 00 00 00 8A 0A 17 9A A0 A1 15 10 8A 00 18 F1 04 00 04 00 5A 00 00 00 1A F3 18 9A A0 A1 15 10 90 00 BE 02 04 00 04 00 1E 00 00 00 C7 20 1C 9A A0 A1 15 10 8B 00 18 11 01 00 01 00 1E 00 00 00 37 09 1E 9A A0 A1 15 10 8C 00 18 21 06 00 07 00 1E 00 00 00 B0 93 20 9A A0 A1 15 10 90 00 BE 32 04 00 04 00 1E 00 00 00 C5 BE 23 9A A0 A1 15 10 8A 00 18 41 04 00 04 00 1E 00 00 00 53 A7 25 9A A0 A1 15 10 90 00 BE 52 01 00 04 00 5A 00 00 00 86 E8 2D 9A A0 A1 15 10 8B 00 18 61 01 00 01 00 5A 00 00 00 5E D2 2F 9A A0 A1 15 10 8C 00 18 71 02 00 06 00 5A 00 00 00 F1 5C 32 9A A0 A1 15 10 90 00 BE 82 01 00 04 00 5A 00 00 00 DE 87 35 9A A0 A1 15 10 8A 00 18 91 01 00 04 00 5A 00 00 00 66 70 37 9A A0 A1 15 10 90 00 BE A2 01 00 04 00 1E 00 00 00 C9 9C 3A 9A A0 A1 15 10 8B 00 18 B1 01 00 01 00 1E 00 00 00 D6 86 3C 9A A0 A1 15 10 8C 00 18 C1 02 00 06 00 1E 00 00 00 30 11 3F 9A 

2019-08-24 00:15:29  FW_Log_Data:0F 00 00 00 A0 A1 15 10 90 00 BE D2 01 00 04 00 1E 00 00 00 78 3C 42 9A A0 A1 15 10 8A 00 18 E1 01 00 04 00 1E 00 00 00 AD 24 44 9A A0 A1 15 10 92 02 BE F2 01 00 04 00 1E 00 00 00 FD 65 4C 9A A0 A1 15 10 92 02 BE 02 04 00 04 00 1E 00 00 00 F6 92 4F 9A A0 A1 14 10 AD 00 BF 12 02 00 00 00 00 00 00 00 A1 2B 51 9A A0 01 15 10 AF 00 30 22 02 00 01 00 00 00 00 00 8A 39 51 9A A0 01 15 10 88 02 34 32 00 00 00 00 00 00 00 00 52 3C 51 9A A0 01 0B 10 38 00 9E 40 00 1B 00 00 80 41 FF 3F 3D 44 51 9A A0 21 17 10 8E 02 66 52 02 00 05 00 00 00 00 00 77 4C 51 9A A0 21 14 10 AD 00 BF 62 03 00 00 00 00 00 00 00 7B 2C 54 9A A0 21 15 10 90 00 BE 72 01 00 05 00 3C 00 00 00 33 EA 5C 9A A0 21 15 10 8B 00 18 81 01 00 01 00 3C 00 00 00 9C D2 5E 9A A0 21 15 10 8C 00 18 91 01 00 09 00 3C 00 00 00 39 00 62 9A A0 21 15 10 90 00 BE A2 01 00 05 00 3C 00 00 00 88 2C 65 9A A0 21 15 10 8A 00 18 B1 01 00 05 00 3C 00 00 00 E8 14 67 9A A0 21 15 10 90 00 BE C2 01 00 05 00 1E 00 00 00 94 42 6A 9A A0 21 15 10 8B 00 18 D1 01 00 01 00 1E 00 00 00 38 2B 6C 9A A0 21 15 10 8C 00 18 E1 01 00 09 00 1E 00 00 00 6C B4 6E 9A A0 21 15 10 90 00 BE F2 0 [... cut ...]
2019-08-24 00:15:50  FW_Log_Data:0F 00 00 00 A0 A1 1D 10 97 00 44 C4 03 00 01 00 10 00 00 00 B0 9B E2 BB A0 01 0D 10 5B 00 1F D2 2B 00 0F 00 00 00 00 00 09 A7 E2 BB A0 01 0D 10 10 01 24 E2 00 00 00 00 00 00 00 00 DF A9 E2 BB 

2019-08-24 00:15:51  FW_Log_Data:0F 00 00 00 A0 01 0D 10 0F 01 24 F2 00 00 F4 01 00 00 00 00 38 6C FB BB A0 A1 1D 10 97 00 44 04 03 00 01 00 10 00 00 00 18 34 14 BC A0 01 0D 10 10 01 24 12 00 00 00 00 00 00 00 00 85 3F 14 BC A0 A1 1D 10 97 00 44 24 03 00 01 00 30 00 00 00 D0 19 30 BC A0 01 0D 10 30 01 24 32 00 00 00 00 00 00 00 00 61 24 30 BC A0 A1 1D 10 97 00 44 44 03 00 01 00 10 00 00 00 A0 17 4C BC A0 01 0D 10 10 01 24 52 00 00 00 00 00 00 00 00 0D 23 4C BC

pwalczykk, 2019-09-06:
And a corresponding experiment log:

==========================
 New attempt: 673/10000
 Date: Fri Aug 23 22:15:18 UTC 2019
=== realsense-kill-all-containers ===
Available Realsense cameras:
    Serial Number                 : 	811112060204
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
Sending SIGKILL to all containers
d0adf84d13c3
Killing containers finished. Available Realsense cameras:
    Serial Number                 : 	811112060204
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
All cameras are OK
==========================
 New attempt: 674/10000
 Date: Fri Aug 23 22:15:51 UTC 2019
=== realsense-kill-all-containers ===
Available Realsense cameras:
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
Sending SIGKILL to all containers
e8525f734b5d
Killing containers finished. Available Realsense cameras:
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
MISSING CAMERA! XHCI RESET IN 30 SECONDS!
=== realsense-restart-xhci ===
Available Realsense cameras:
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
Unbinding...
echo 0000:19:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1a:00.0 > /sys/bus/pci/drivers/xhci_hc:?d/unbind
echo 0000:1b:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
echo 0000:1c:00.0 > /sys/bus/pci/drivers/xhci_hcd/unbind
Binding...
echo 0000:19:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1a:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1b:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
echo 0000:1c:00.0 > /sys/bus/pci/drivers/xhci_hcd/bind
XHCI restart finished. Available Realsense cameras:
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
MISSING CAMERA! HARD RESET IN 30 SECONDS!
=== realsense-restart-hard ===
date: extra operand ‘3’
Try 'date --help' for more information.
Shutdown scheduled for Sat 2019-08-24 00:17:35 CEST, use 'shutdown -c' to cancel.
==========================
 New attempt: 675/10000
 Date: Fri Aug 23 22:17:35 UTC 2019
=== realsense-kill-all-containers ===
Available Realsense cameras:
    Serial Number                 : 	811112060204
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
Sending SIGKILL to all containers
e52349ab52af
Killing containers finished. Available Realsense cameras:
    Serial Number                 : 	811112060204
    Serial Number                 : 	802212061216
    Serial Number                 : 	802212060668
    Serial Number                 : 	811112060576
All cameras are OK
==========================

RealSenseCustomerSupport, 2020-06-16:
----------------------------------------------

@pwalczykk Sorry for late reply. Is the issue still existing for you? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

@pwalczykk Could you please update? Thanks!

pwalczykk, 2020-06-22:
@RealSenseCustomerSupport I haven't continued this experiment due to a lack of responses. Even with firmware 5.12.3 I can "break" camera by terminating Docker container in which ROS driver for the camera is streaming. However, I have learned to live with it. It is enough to shut down PC and set a wakealarm which will boot PC sometime later. If the camera does not have a power supply for a few seconds, it usually recovers after boot. Instruction on using wakealarm can be found here: https://www.mythtv.org/wiki/ACPI_Wakeup#Using_.2Fsys.2Fclass.2Frtc.2Frtc0.2Fwakealarm

RealSenseCustomerSupport, 2020-06-29:
----------------------------------------------

@pwalczykk Thanks for your reply and sharing! Will close this issue accordingly. Please feel free to create new one if you have any other questions or issues. Thanks!

closed 2020-06-29 by RealSenseCustomerSupport