realsenseai/librealsense #4553: How to remove ghost noise from realsense D435i ?

MrYelameli, 2019-08-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 

| Operating System & Version |                   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.15.0-55| 
| Platform                                 | PC  |

The point cloud generated by realsense have much noise, the points are being dragged and looking as ghost, probably popularly known as "ghost noise". We tried some filters without success. It is difficult to identify the objects in the point clouds. So is there any specific filtering technique or any other method which removes this kind of noises. For better understanding I have added the images. 
[image]

[image]

[image]

It can be seen from the images that, person standing in the image is being seen as a ghost (point clouds are dragged). It is not only person but many other objects too. How to filter it?

MartyG-RealSense, 2019-08-02:
I recall a past case where the point cloud was being elongated (which might be similiar to the 'dragged' effect that you describe).

https://github.com/IntelRealSense/realsense-ros/issues/507

A user in a different case also experienced this effect when viewing a banana from the side.

[image]

MrYelameli, 2019-08-05:
@MartyG-RealSense  thank you for your reply. But where is the solution? The link you specified does not specify any technique or solution. Thank you.

MartyG-RealSense, 2019-08-05:
I tracked down a link about the case of the user with the twisted banana, where advice is given by a member of the RealSense team.

https://github.com/IntelRealSense/librealsense/issues/3741

Garand0o0, 2019-08-08:
Hi MrYelameli. I had the same problem. Do you have a solution? Thank you.

unagiHuman, 2019-08-19:
There is a way to reduce noise using post-processing-filters.
For example, if you use a decision-filter, the resolution will be low, but the noise will be greatly reduced.

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

Just to add to the great suggestions from MartyG and unagiHuman, below is the up-to-date calibration tool and white paper for getting the best performance from the RealSense D400 cameras.

Dynamic Calibration Tool:
https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

Tuning depth cameras for best performance:
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

Please let us know if further assistance is needed. Thank you.

closed 2019-09-16 by RealSenseCustomerSupport