realsenseai/librealsense #4554: A factor is not changed.

Mustardsauce, 2019-08-02:
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.24, 2.25  | 
| Language                            |  C++  | 
| Segment			|  others  | 

Hello, I saw this white paper.
(https://dev.intelrealsense.com/docs/white-paper-subpixel-linearity-improvement-for-intel-realsense-depth-cameras)

I tried to change A factor using C++ language or .json file or Realsense View.
But It's not changed. How can I change this A factor value?

ev-mp, 2019-08-04:
@Mustardsauce , A-factor functionality [requires FW 5.11.9+](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements-2). 
I believe a compatible version will be released soon though I don't have a concrete info. @RealSense-Customer-Engineering , can you comment ?

MartyG-RealSense, 2019-08-27:
@Mustardsauce  The new Development Firmware 5.11.11.100 is available for download from the firmware releases page.  Among its updates are improved depth linearity.  Details of the additions in this firmware can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360034784453

dorodnic, 2019-09-03:
Hope it works well now. Please feel free to open a new issue if it does not

closed 2019-09-03 by dorodnic