realsenseai/librealsense #4555: Announcement: Introducing  the AAEON UP Squared Robomaker Dev Kit

MartyG-RealSense, 2019-08-02:
Hi everyone,

Intel have published an article about the 'AAEON UP Squared Robomaker Dev Kit', a bundle containing:

- A RealSense D435i camera

- Up Squared development board (more powerful than the original Up Board in the Intel Click store)

- UP AI Core X module powered by the Intel Movidius Myriad X VPU

- M.2 2230 WiFi Kit for UP Squared metal Chassis

- Software package with AWS RoboMaker integrated

*****

The Intel article about the Robomaker Dev Kit can be found here:

https://www.intelrealsense.com/robomaker/

GOBish, 2019-08-04:
How needed is the UP AI Core X module if, for example, I would be doing person tracking? Currently I am experimenting with NUItrack to do this.

MartyG-RealSense, 2019-08-04:
You shouldn't need this module at all to do person tracking with Nuitrack.  A depth camera such as the D435i should suffice.   Intel's article implies that the module will improve performance of computer vision applications of stereo depth cameras like the RealSense 400 Series though.

Nuitrack has a new tracking product coming called Nuitrack AI that claims to provide superior tracking performance.

[image]

closed 2019-08-12 by dorodnic