realsenseai/librealsense #4556: SR305 not recognized in Raspbian

evang2v, 2019-08-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   |  | 
| Operating System & Version |   Raspbian 9.9  | 
| Kernel Version (Linux Only)    |  4.19.57                                        | 
| Platform                                 | Raspberry Pi 3|
| SDK Version                            |  2.25                          | 

### Issue Description
-I followed [this guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md) for installation.
-RealSense Viewer does not recognize the SR305 on my Raspberry Pi 3.
-I also have a D415 which is correctly recognized and works fine (after updating to the latest firmware). I am not trying to use them at the same time.
-The SR305 correctly installs on a Windows machine (but I have not tested it with RealSense Viewer). 
-The SR305 does not show up in lsusb (but of course the D415 does).

Any advice on getting the SR305 recognized would be appreciated.

MartyG-RealSense, 2019-08-02:
Have you patched the uvcvideo driver, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

evang2v, 2019-08-02:
I have not. Could you please give me a hint how to do it on Raspbian? 

`patch-realsense-ubuntu-lts.sh` fails with error 
>E: Package 'linux-headers-generic' has no installation candidate.

MartyG-RealSense, 2019-08-02:
Raspbian is a bit outside of my experience, though I have handled a couple of cases involving it.  A RealSense community member has contributed a guide to installing Librealsense on Raspbian. It is aimed at installation on a Raspberry Pi 3 or 3+ and D435 camera.  that may provide some useful insights.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

evang2v, 2019-08-02:
Yes, this is exactly the guide I followed, which allows my D415 to work. There is no mention of uvc patching in this guide as far as I can see.

MartyG-RealSense, 2019-08-02:
I see that the Raspbian instructions use the CMake flag **-DFORCE_LIBUVC=true** ... it bypases video4linux, avoiding the need to patch the kernel.  so that would explain why patching uvcvideo largely isn't mentioned.

https://github.com/IntelRealSense/librealsense/issues/1132#issuecomment-363572666

ev-mp, 2019-08-04:
@evang2v , SR305 is a [USB3 product](https://www.intelrealsense.com/depth-camera-sr305/), which makes it incompatible with Raspberry Pi 3 that has only USB2 ports ([Link](https://www.raspberrypi.org/magpi/raspberry-pi-3-specs-benchmarks/)).

In case you're developing for Raspbian then you may consider the newer Raspberry Pi 4 model which received USB3 ports.

MartyG-RealSense, 2019-08-04:
Oh yes, of course SR305 would not have a USB2 mode, since SR300 (the camera technology it is based on) is USB 3!  @evang2v my apologies for not thinking of that!

evang2v, 2019-08-06:
All good, all clear! Thanks for the help.

Unless I'm missing something, there is no way to tell from the product pages that the D415 supports USB2 and the SR305 does not, correct?

closed 2019-08-06 by evang2v