realsenseai/librealsense #4557: undefined symbol: libusb_get_port_numbers

hosseinpg, 2019-08-02:
Hello,
I have a D435, trying to run my first project in Ubuntu 16.04. I have installed everything and there is no error when I try to make my project. But when I try to run the generated file in build dir while the camera is plugged in, I get following error: 
./scanner: symbol lookup error: /usr/lib/x86_64-linux-gnu/librealsense2.so.2.24: undefined symbol: libusb_get_port_numbers

FYI, I have this line in my cmake file: target_link_libraries(scanner /usr/lib/x86_64-linux-gnu/libusb-1.0.so). Removing this line leads me to get other linking errors. Regarding my src file, I'm trying to compile rs-pcl project. 

Any help is appreciated.

hosseinpg, 2019-08-04:
Problem solved by forcing the cmake to find right version of libusb.

closed 2019-08-04 by hosseinpg

wf-hahaha, 2020-03-28:
@hosseinpg hello，Did you solve it? how to solve it？