realsenseai/librealsense #4558: No fonts in realsense-viewer GUI after compiling from source on Mac OS 10.13.6

alifeinbinary, 2019-08-02:
Hello everyone, 
I'm having an issue when compiling from source on a Macbook Pro (late 2011) running Mac OS 10.13.6. XCode is installed and running properly. This is what the realsense-viewer GUI looks like after compiling. I've tried recompiling a number of times but the issue persists. I'm running 10.13.6 on my desktop, as well, but don't have this issue on that machine. The only flag I'm appending to `cmake` is `-DBUILD_PYTHON_BINDINGS:bool=true`
[image]

the last few lines after running `sudo make install` might elude to the what's causing the issue. Any help would be much appreciated as I'm doing most of my development on the road these days. Cheers 
[image]

dorodnic, 2019-08-04:
Hi @alifeinbinary 
This sometimes occurs when there is not enough texture memory for the font atlas.
Try hard-coding the value: [`OVERSAMPLE`](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L117) to 1

alifeinbinary, 2019-08-05:
@dorodnic where do I write this value? at the top of realsense-viewer.cpp?

dorodnic, 2019-08-06:
You need to modify model-views.cpp line 117 from:
`const int OVERSAMPLE = config_file::instance().get(configurations::performance::font_oversample);`
to...
`const int OVERSAMPLE = 1;`
Save and rebuild the Viewer.

Alternatively, you could edit file `~/.realsense-config.json`, adding line:
`  "performance.font_oversample": "1",`
anywhere in the file. This method would not require rebuilding the Viewer.

alifeinbinary, 2019-08-15:
@dorodnic the first method you suggested worked for me, the second one did not. Just for future reference for anyone checking this thread. Thanks for your help!

closed 2019-08-15 by alifeinbinary

lacandelaria, 2020-10-12:
Hello @alifeinbinary!
I had managed to download and open the realsense Viewer on my Mac, and I had your same problem about fonts. Therefore I decided to delete it and went on using a PC.
Now I need to download it again on another Mac for a project, but I can't just do it anymore. It seems like there is no way to do that by following those steps on the Intel site, as if in a few months the libraries had changed or something. 
Can I ask you how you did it? The errors I see are all about the CMakeLists.txt file, and I don't really know how to fix them.