realsenseai/librealsense #4559: Resolution guesser

abielalgravez, 2019-08-03:
[image]

Hi, I'm getting my point cloud, although it only shows it from certain range. After that, it just replicates the last point and extends it as shown in the image. Anyone knows how to solve this?

ev-mp, 2019-08-04:
@abielalgravez , apparently you're using D435(i), and this phenomenon is due to the differences in Depth and RGB field of view angles. The RGB FOV is much narrower than depth so the peripheral pointcloud points do not have a genuine texture that can be applied.
Duplicate of #3106, #2355, #2237

abielalgravez, 2019-08-04:
@ev-mp Thank you for your response. So, is there any way to omit those points that don't have an RGB pair so that the phenomenon is not displayed?

ev-mp, 2019-08-04:
The pointcloud does not define how the pixels are drawn-  it provides a map of normalized [-1...1] texture coordinates that apply to each depth point. It is up to the renderer to select and apply the proper texel. In case of `realsense-viewer` this logic takes place in `pointcloud-gl.cpp` shader.
Note that there is an alternative implementation in CPU in `example.hpp` file where you can inspect both the vertex and the uv coordinate before sending them to opengl pipe.
```c++
        if (vertices[i].z)
        {
            // upload the point and texture coordinates only for points we have depth data for
            (Add you code here to filter out depth point with no proper texture available ...

            glVertex3fv(vertices[i]);
            glTexCoord2fv(tex_coords[i]);
        }
```
This implementation is used in another demo - `rs-pointcloud`

abielalgravez, 2019-08-05:
Sounds good! Thank you. I will give it a look.

closed 2019-08-05 by abielalgravez