realsenseai/librealsense #4563: realsense SDK 2.0 install error

seunghak-kim, 2019-08-04:
I got an error while installing realsense SDK 2.0 on Ubuntu 18.04 LTS.

https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.mdhttps://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md

I followed the installation by following the instructions on this site.
However, as soon as I started the installation, I got an error in registering the server's public key.

wireless@ubuntu:~$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
[sudo] password for wireless: 
Executing: /tmp/apt-key-gpghome.3fhmz5IoEy/gpg.1.sh --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

I did not understand this error. Please help me.

ev-mp, 2019-08-04:
@seunghak-kim , the system reports that the public key has already been obtained, probably from the previous installations. I don't see any errors - did you try to continue the installation and what were the results ?

seunghak-kim, 2019-08-04:
Installation seems to be done. Because the realsense-viewer, the D435 camera image is displayed well. However, I think that the registration of public key is not installed completely because of error.

ev-mp, 2019-08-04:
@seunghak-kim ,  since you haven't fill in the issue template, I cannot tell which installation path you've followed (Debians or Building from source), and it is hard to be precise.
Nevertheless -  had the registration failed you wouldn't be able to use `sudo apt install librealsense2...` command which is used in the latter. The reported line
>gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" **not changed**

suggests that the key has been already/previously installed and there is no need for it to be updated. 

In the former case registration of the public key is not necessary.
In any case the attached log does not include error messages so no further actions are required.

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

@seunghak-kim Could you please provide more information and logs so that we can look into it further? Thanks!

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

@seunghak-kim Could you please update more information and the logs? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-09-16:
----------------------------------------------

@seunghak-kim  Since the issue resolve, will close it accordingly. Thanks!

closed 2019-09-16 by RealSenseCustomerSupport