realsenseai/librealsense #4564: How to use RealSense in .Net Core Native?

smirkingman, 2019-08-04:
I have a RealSense .Net Core project that works fine.

When I generate a single-EXE file with the ILC (https://devblogs.microsoft.com/dotnet/the-net-native-tool-chain/) it fails when creating the config with this error:

 System.Exception: Method '[Intel.RealSense]Intel.RealSense.NativeMethods.rs2_create_config(object&)' requires marshalling that is not yet supported by this compiler.
  at Intel.RealSense.NativeMethods.rs2_create_config(Object&) + 0x34
  at Intel.RealSense.Config..ctor() + 0x23
  at RealSenseStreamer.BaseStreamer.FrameGrabber() + 0x5d (<- my code 'New config')

Is .Net Native supported?

dorodnic, 2019-09-03:
> Is .Net Native supported?

No, not officially. 
Ultimately, you have to include the native `realsense2.dll` for this to work, I don't see any way around it.

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

Hi @smirkingman,

Do you need any further clarifications regarding this topic?

smirkingman, 2019-09-06:
Hi Sergey,
thanks for the follow-up. I don't understand your suggestion:

> Ultimately, you have to include the native realsense2.dll for this to work, I don't see any way around it.

When I publish as a netcoreapp2.1, realsense2.dll gets copied to the publish directory (so presumably the build recognised tghe dependency) and the program starts but fails with 

> requires marshalling that is not yet supported by this compiler.

How can I workaround this?

aalmada, 2019-10-17:
@smirkingman In case there is no `corert` documentation about marshaling limitations, I suggest checking its source code to find why this exception is thrown: https://github.com/dotnet/corert/blob/71f811248a4e3696d35fa42b7e4884ab66c22d5e/src/Common/src/TypeSystem/IL/Stubs/PInvokeILEmitter.cs#L355

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

Hi @smirkingman,

Did you have a chance to try out @aalmada's suggestion?
At this stage of development .NET Native precompilation technology is not supported by Intel RealSense SDK 2.0. We are working on adding support for it but there is not a schedule yet.

Thank you!

smirkingman, 2019-10-27:
@aalmada's suggestion indeed answers my question, the header of PInvokeILEmitter.cs says 

> Provides method bodies for PInvoke methods
> This by no means intends to provide full PInvoke support. The intended use of this is to
> a) prevent calls getting generated to targets that require a full marshaller
> (this compiler doesn't provide that), and b) offer a hand in some very simple marshalling
> situations (but support for this part might go away as the product matures).

I was just trying out the precompilation to see if it was an envisageable avenue; if support is planned, I'm happy to wait.

Many thanks, the support here is exceptional, expert and diligent :)

closed 2019-10-27 by smirkingman