realsenseai/librealsense #4565: Raspbian GNU/Linux 10 (buster) compile rs-software-device

bgilreath, 2019-08-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  4.19.58-v7l+                            | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |   2.25                          | 
| Language                            |  {C/opencv/python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Finished prerequisites of target file 'examples/software-device/CMakeFiles/rs-software-device.dir/build'.
The prerequisites of 'examples/software-device/CMakeFiles/rs-software-device.dir/build' are being made.
Live child 0x121af68 (examples/software-device/rs-software-device) PID 2773 
[ 57%] Linking CXX executable rs-software-device
Reaping winning child 0x121af68 PID 2773 
Live child 0x121af68 (examples/software-device/rs-software-device) PID 2775 
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
Reaping losing child 0x13e4590 PID 2762 
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:91: examples/hello-realsense/rs-hello-realsense] Error 1
Removing child 0x13e4590 PID 2762 from chain.
Need a job token; we have children
Live child 0x198ff68 (examples/software-device/CMakeFiles/rs-software-device.dir/all) PID 2772 
Live child 0x198f8c0 (examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all) PID 2758 
Reaping losing child 0x198f8c0 PID 2758 
make[1]: *** [CMakeFiles/Makefile2:455: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
Removing child 0x198f8c0 PID 2758 from chain.
Released token for child 0x198f8c0 (examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all).
make[1]: *** Waiting for unfinished jobs....
Live child 0x198ff68 (examples/software-device/CMakeFiles/rs-software-device.dir/all) PID 2772 
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
Reaping losing child 0x121af68 PID 2775 
make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/build.make:91: examples/software-device/rs-software-device] Error 1
Removing child 0x121af68 PID 2775 from chain.
Reaping losing child 0x198ff68 PID 2772 
make[1]: *** [CMakeFiles/Makefile2:511: examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
Removing child 0x198ff68 PID 2772 from chain.
Reaping losing child 0xb45de0 PID 2713 
make: *** [Makefile:130: all] Error 2
Removing child 0xb45de0 PID 2713 from chain.

filipematosinov, 2019-08-05:
Same problem here ;(

ev-mp, 2019-08-05:
@bgilreath , thanks for reporting.
Apparently this is a requirement specific to Raspbian builds ([Link1](https://github.com/alexa/avs-device-sdk/issues/380) ,[Link2](https://github.com/pytorch/pytorch/issues/22898)).
We need to add the fix to the SDK.
In the meanwhile you can apply the fix from the links locally
> set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} **-latomic**")

Xxjojo, 2019-08-27:
@ev-mp hey there,
I have the same problem.
Where in the CMakelist.txt does I have to put the line?

bgilreath, 2019-09-02:
@ev-mp Thanks! The fix you suggested did fix my issue.

closed 2019-09-02 by bgilreath

coriolan-cs, 2019-09-13:
Found the solution, I've got the Realsense working on my RPI4 now.

w3bm0nk3y, 2019-10-06:
Was the final solution to use the libuvc-backend installation instructions and update CMakeLists.txt with set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic") ?

sarathynp, 2020-02-08:
I am trying to install on Raspberry PI 4.
I am following the instructions from https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

My installation wnet all the way up to 57% and failed as 
 ./libuvc_installation.sh

 57%] Linking CXX shared library librealsense2-gl.so
[ 57%] Built target realsense2-gl
make: *** [Makefile:130: all] Error 2

Any help will be appreciated.
Thanks
Sarathy