realsenseai/librealsense #4569: Realsense viewer not working from version 2.18 and above

ArjanVroegop, 2019-08-05:
Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 } | 
| Firmware Version                   | (5.11.6.250 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                       | 
| Platform                                 | Notebook  |
| SDK Version                            |  2.18 and above }                          | 
| Language                            | N/A }                          | 
| Segment			|  {N/A                        | 

### Issue Description
Hi everyone,

I'm having a problem with the latest versions of the Realsense Viewer (2.18 and above). From this version, the viewer does not launch anymore. Version 2.17 works fine. I've found some topics on the same issue, for all of those the solution was to update the graphics driver. However, this doesn't work for me, updating the graphics driver did not help.

I have a Lenovo P51 notebook running Windows 10. This notebook has a Intel Graphics P630 (driver 26.20.100.6913, latest version supported by Lenovo) and a Nvidia Quadro M2200 (driver 25.21.14.2545).

When I launch the viewer, nothing seems to happen. It is possible to run the example programs from the latest SDK.

Does anyone have an idea what I could do to make the newest viewer working?

copiousOne, 2019-08-10:
I am having the same issue, Lenovo P52s.  Updated graphics card Intel UHD 620.

ev-mp, 2019-08-18:
A recent report by user user suggest that this could be due a non-standard/missing windows path that Viewer uses to store configuration/log files rather than video driver. The most distinctive detail is that LRS version prior to 2.18 work properly, as in this case
https://github.com/IntelRealSense/librealsense/issues/3576#issuecomment-521505507

Can you check/share the path for `C:\Users<UserName>\Documents`  , and whether it includes spaces/non-utf characters ?

ArjanVroegop, 2019-08-29:
Hi,
I'm sorry for the late reply, I was off work for a while. I'll have a look at this, this could indeed be the problem because our organisation does some weird stuff with the Windows user profiles. I'll let you know

ArjanVroegop, 2019-08-30:
I've found a solution. In our organisation the Documents folder is not local but somewhere on the network. I've created the folder C:\users\<username>\Documents and now the viewer works (version 2.27.0)

closed 2019-09-19 by ev-mp