realsenseai/librealsense #4571: v2.25 presets not working

NXTwoThou, 2019-08-05:
Camera Model: D415
Firmware Version: 5.11.11.100
Operating System & Version: Win 10 Build 18950
SDK Version: 2.25
Language: C#
Segment: 3D scanning

Just got everything updated to latest versions.  When I pull up the RealSense viewer if I turn on sereo module it'll display.  If I try and select any of the presets(or load my custom one), realsense viewer freezes.  

[image]

Killing the app and trying to start it back up gives a "Backend in rs2_create_device(info_list:00000235D
BFB86C0, index:0):
MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x800706be: "The remote 
procedure call failed."" error.

[image]

Unplugging camera and plugging it back in can get me back up and running, only without presets.  Presets are kind of critical as my camera is located as close as it can get to the object.

[image]

dorodnic, 2019-08-06:
Hi @NXTwoThou 
We are getting more and more reports about "MFCreateDeviceSource" error, my initial impression is that something in the media subsystem was broken in the latest Windows update.
Once we have sufficient information about this problem, we'll try to provide a reasonable work-around, but there is a chance we are dealing with an OS bug...

NXTwoThou, 2019-08-06:
Please let me know if there's anything I can do to assist.  I have a D415 and a SR305 here.  Note, without using a preset, it seems to work properly.

NXTwoThou, 2019-08-07:
Note, I tried the HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows Media Foundation\Platform DWORD EnableFrameServerMode to 0 change that was mentioned in a different posting as fixing their issue, didn't help here.  Same issue where no preset the camera works, any preset, locks up.

NXTwoThou, 2019-08-07:
Just updated to build 18956, same issue.

NXTwoThou, 2019-08-12:
I figured I'd update the logging with 18956 just in case it is a windows error more info might come out of the logs...

I loaded the viewer up, clicked Info so it'd display it on screen, left preset at "Custom", turned on the stereo module, it displayed what I expected.  

[image]

I then turned off the stereo module, selected "Default" for the preset and turned the stereo module.  Got a wait cursor and eventually the app went not responding.

[image]

Here's the debug log for this sequence of events.

[librealsense.log](https://github.com/IntelRealSense/librealsense/files/3492608/librealsense.log)

NXTwoThou, 2019-08-17:
FIXED with build 18963!  It was a Windows 10 issue.
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/3512013/librealsense.log)
[image]

closed 2019-08-17 by NXTwoThou