realsenseai/librealsense #4573: Feature Request: Downscale on device to save bandwidth

TyTruGolf, 2019-08-05:
I've noticed that the depth quality of 848x480 with decimation 2, is DRAMATICALLY better than 424x240 with no decimation. 

As someone who's running multiple cameras at 90fps, its frustrating that I have to take such a huge bandwidth hit for a resolution that I don't need just to get the level of quality that I do. 

If the camera could downscale on the device, than I would be able to run multiple devices reliably and get high quality information.

Personally, I would rather the 424x240 resolution was just automatically optimal resolution down scaled, as it does not make a lot of sense to me why it would be much worse quality. (I mean, I've read into it, so I understand now, but as a user I would expect the camera to do what it needs to do to give me the best quality it can for each resolution)

As an alternative, it would be good to have an option to downscale, much like how we enable auto-exposure or set the laser level.

agrunnet, 2019-08-05:
@TyTruGolf you are right. It would be better to have separated capture resolution from output resolution and allow for downscaling on the chip. Unfortunately we omitted to do it. It will certainly be in a next version of the chip. For now you have to downsample yourself as per our what paper on post processing.

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

@TyTruGolf We will do this in next ASIC/chip.
Can we close this ticket for now?

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@TyTruGolf
We will tape out next ASIC by the end of 2020

TyTruGolf, 2019-09-23:
Thanks, you can go ahead and close the ticket.

TyTruGolf, 2019-09-23:
Oh, yeah, we wanted to know if there's any kind of beta program for testing this out when you're in development. We'd like to be able to track a golf ball going 200mph and that would be a good test case for your engineers and potentially good showcase for the capabilities.

agrunnet, 2019-09-23:
We are actually working on a high speed 400fps mode that we hope to launch soon. It will have resolution 848x100 and will work with the D435 and D435i.  I believe this should work for you.

TyTruGolf, 2019-09-23:
Hey, that will be pretty cool! 

Not to beat a dead horse here, but will the 848x100 resolution have the same issues that 424x240 has in that it won't have near the same depth quality as 848x480 downsampled?

agrunnet, 2019-09-23:
They horse is alive. The quality will be best possible for D435. It is same as 848x480.

TyTruGolf, 2019-09-24:
Oh wow, that's awesome!
I'm not sure 100px vertical resolution is enough to pick up a golf ball, is there any chance we could get an 848x200 @200fps? That would really solve all of my problems!

Also, I really appreciate all of the time you guys are putting into this. It's an amazing device!

TyTruGolf, 2019-09-25:
Actually, to the original point of this post, if only horizontal resolution is needed for depth quality, could we get an 848x240 resolution? That'll give me the same quality I'm getting now at half the bandwidth!

(and yes, 848x240@180fps on existing hardware would be AMAZING)

agrunnet, 2019-09-25:
It is a good ask. 
However in the meantime probably good to first check what we have. Try 90fps first. Set exposure low. See what issues you see. Then get the 400fps version to try.

TyTruGolf, 2019-09-25:
I loaded up 848x100, but it looks like it's just 848x480 cropped down to 848x100. Is there a way to get the high frame rate and keep the high vertical FOV?

3ms exposure has been our safest universal. 2.5ms works for most setups and 2ms does OK. 

Thanks for all the quick responses. I really appreciate how responsive you guys have been as well as all the insight you've been able to provide.

agrunnet, 2019-09-25:
Unfortunately we are limited by the sensor and the bandwidth. By cropping down the vertical FOV the global shutter imager is HW windowing the image and we can cycle through it faster, without increasing bandwidth. So bandwidth is roughly equivalent to 848x480 at 90fps. But you can use multiple cameras to cover the FOV you want ;)

TyTruGolf, 2019-09-25:
Oh ok.  So in order to keep the same quality but increase speed, you have to have the sensor capture at 848x480 and than only process a part of it. You can't change the sensor to capture the whole frame at 848x240 (similar to how we can tell it to capture at 424x240) and get the same HQ benefits.

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Is there any more question about this ticket?

Thanks

TyTruGolf, 2019-10-09:
Nope, that's everything. 
Thanks so much for all of the info and I'm looking forward to the future updates!

closed 2019-10-09 by TyTruGolf