realsenseai/librealsense #4576: rs-align-advanced Sample Performance issue

abielalgravez, 2019-08-05:
Hello,

I'm using the rs-align-advanced Sample (unmodified) and it performs very slowly in comparison to the other samples I've run. I'm using the Realsense D435 camera and running it on a new Acer Nitro 5 gaming computer. Any idea why this is happening?

abielalgravez, 2019-08-05:
I narrowed down the search by tracking the processing time. Turns out the line:
        auto processed = align.process(frameset);
is the one causing the program to delay around 0.8 seconds every time is run. Any ideas on how to speed it up?

abielalgravez, 2019-08-05:
I tracked it down to the rs_processing.hpp file. In it the error handler:

```
            rs2_error* e = nullptr;
            rs2_process_frame(get(), ptr, &e);
            error::handle(e);
```
is the one delaying the code. I can't just comment it. How can I proceed?

dorodnic, 2019-08-06:
Good chance code is compiled as Debug, no processing should take 0.8 seconds
Please make sure to build as Release

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

abielalgravez, 2019-08-06:
It worked! Wonderful! Thank you very much!

closed 2019-08-06 by abielalgravez