realsenseai/librealsense #4577: How can I get the depth values from the raw image/png images in centimeters/millimeters?  I want to get the high precision values. Not by just converting the meters into centimeters.

ankith001, 2019-08-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-08-05:
@ankith001 , to convert the raw depth data (uint16_t) into metric values you need to multiply those by the `depth_scale` value ,e.g. [Link](https://github.com/IntelRealSense/librealsense/issues/3865#issue-438201126):
>depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()

This multiplication is also performed internally in [`rs-hello-realsense`](https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense) example

You  cannot extract depth values from the exported PNG (colored) depth images.  The color mapping is a one-way function due to histogram equalization.

RealSenseCustomerSupport, 2019-08-29:
----------------------------------------------

@ankith001, can we close this ticket if there is no further question?

closed 2019-09-05 by RealSenseCustomerSupport