realsenseai/librealsense #4578: USB error T265 jetson TX2

JvanZijl, 2019-08-05:
Dear all,

For some time I have been trying to use the T265 on the Jetson TX2. I have installed Jetpack 3.2.1 on the Jetson and have built realsense using TM2. The T265 is recognized when running realsense, but when I turn on the tracking module however, the camera turns on for a second but then crashes with the following error log:

20:36:45.681 [4374] [E] Device-DFC0: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 196608
20:36:45.882 [4360] [E] Device-DFC0: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
20:36:45.882 [4360] [E] Device-DFC0: Entered state [ERROR_STATE]
 05/08 20:36:46,249 ERROR [547526176768] (types.h:243) Failed to query T2xx tracking temperature option
 05/08 20:36:46,383 ERROR [547526176768] (types.h:243) Failed to query T2xx tracking temperature option

When I disable the 2 fisheye camera streams in realsense viewer I am able to succesfully view the accelerations and such, but as soon as I turn the video stream on it will crash again.

Some points I feel i should mention:

- I am using a powered USB 3.0 hub
- I have used the Cmake option with TM2 enabled
- I have tried it with both a patched and unpatched kernel

At this point I am running out of ideas but would really like to get this to work on the Jetson TX2. I hope someone can help me.

Kind regards.

peterwatsonnm, 2019-08-07:
@JvanZijl I saw your comment on my post #4592 , unfortunately I'm not great at this technology yet.  It sounds like the T265 is sending more data though your USB than it can handle.  There is some documentation about fish eye data not working on usb 2, meaning that fish eye sends a lot of packets through the cable even if it is a usb 3.  Are you using a usb 3 cable as well, or just the hub?

JvanZijl, 2019-08-07:
@peterwatsonm Thank you for your answer! It was indeed an USB cabling issue. The cable between the T265 and the power hub was fine, but for some reason the Realsense connects through the USB 2.0 bus, instead of the 3.0 bus. This happens even though the Realsense is connected through a powered USB 3.0 hub. Will try to figure out why this is happening. Thanks again!

peterwatsonnm, 2019-08-07:
no worries.  I just spend a while troubleshooting my setup, and it ended up being the usb cable as well.  Good luck with the rest of it!

closed 2019-08-08 by JvanZijl