realsenseai/librealsense #4579: Problem with jupyter distance_to_object Notebook Intel Real Sense

candrescn, 2019-08-06:
Hi, I'm trying to run this example with the guide https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

Perfectly arrive in In [4] and Out [4]

but when arriving at the lines In [5]:

 

colorizer = rs.colorizer ()

colorized_depth = np.asanyarray (colorizer.colorize (depth_frame) .get_data ())

plt.imshow (colorized_depth)

 

throws me an error: np.asanyarray (colorizer.colorize (depth_frame) .get_data ()) RuntimeError: null pointer passed for argument "frame"

 

I'm stuck there and I don't know what to do, maybe you can help me?

@dorodnic i need your help pleaseee

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer | 
| Operating System & Version |  (Ubuntu 16  | 
| Kernel Version (Linux Only)    | 4.14.13)                                         | 
| Platform                                 | UPboard.  |
| SDK Version                            |                         | 
| Language                            | python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          |

candrescn, 2019-08-20:
Hello @dorodnic , you may have some idea how to solve it, I tried to config.enable_stream and run pipeline with parameter config but I still get the same error.

lramati, 2019-09-02:
I couldn't reproduce your issue, but I just opened #4787 to try and fix some other issues. Could you see if the following binder link to my fork works for you?

[image]

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

HI @candrescn,

Did you try the update from @lramati and see how it works?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

HI @candrescn,

I will close this if no additional question.

closed 2019-11-01 by RealSenseCustomerSupport