realsenseai/librealsense #4581: RealSense error calling rs2_create_device 0x800706d9

yuemasy, 2019-08-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435 | 
| Firmware Version                   |  can't be seen | 
| Operating System & Version |    Win 10 ver 14393.2580  |  
| Platform                                 | PC  |
| SDK Version                            |  2.17.0, 2.17.1, 2.25.0                           | 
| Language                            |  { NO }                          | 
| Segment			|  { VR }                          | 

 
I tried both D415 and D435 with the RS viewer. They just can't work on this pc and worked well on other laptop and pc which is also version 14393. I have to use this win10 version.  
When install SDK 2.17.0 and 2.17.1, the viewer showed nothing. Uninstall and reinstall many times still the same result.
When install SDK 2.25.0, it showed "RealSense error calling: backend in rs2_create_device... 0x800706d9", the error code can't be seen completely. I use rs-emumerate-devices.exe to see it.

[image]
[image]

I checked settings>devices>D435 can be seen and shows "connected to USB 3.0"
I checked Device manager>Imaging devices>both RGB and depth camera can be  seen
I checked camera privacy settings is on. 
Thanks a lot. 
@dorodnic @MartyG-RealSense

MartyG-RealSense, 2019-08-06:
I had a case with the MFCreateDeviceSource error recently.  They too had their privacy correctly set to On.  My recollection is that in that case, the error was being caused by antivirus software on their computer.  If the PC the camera does not work on has antivirus, this may account for the problem.  Do you know if this PC has such software, please?

yuemasy, 2019-08-07:
@MartyG-RealSense I couldn't found any antivirus software on that PC. I found the solution just now.
It is the solution that solves win10 camera can't be used.
-  win + R, run 'regedit';
- go to 'HKEY_LOCAL_MACHINE \ SOFTWARE \ Microsoft \ Windows Media Foundation \ Platform'；
- add DWORD(32bit) value ，name as “EnableFrameServerMode”，the default value 0 is ok；
- run RS viewer;
when the viewer open at the first time, an error message displayed at the right top corner but then it disappeared. After that I turn on the camera, both the depth and RGB image can display.
I don't know if there is any problem in the future.
Thanks.
[image]

MartyG-RealSense, 2019-08-07:
I'm very pleased that you found a solution.  Thanks for sharing it with the community!

closed 2019-08-27 by RealSenseCustomerSupport