realsenseai/librealsense #4585: How to install pyrealsense2 in conda environment??

lokanadha27, 2019-08-06:
I'm using Windows 10
D435 camera
using anaconda

tried to install by giving conda install pyrealsense2
it failed.
Please help me out

LLUWE19, 2019-08-08:
Think you might need at least version 3.5 for conda or pip to work

lokanadha27, 2019-08-08:
@LLUWE19 
you mean the latest version of python? I have python 3.7 version.

LLUWE19, 2019-08-08:
Did you install the realsense from source??

LLUWE19, 2019-08-15:
Sorry, you actually need to python 3.6 as the pyrealsense does not support 3.7 yet

RealSenseCustomerSupport, 2020-01-20:
----------------------------------------------

Updated readme to reflect latest status of python version support...

https://github.com/IntelRealSense/librealsense/pull/5679

Related Issues:

https://github.com/IntelRealSense/librealsense/issues/4765
https://github.com/IntelRealSense/librealsense/issues/4515

Will you be needing further assistance with this?

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-04-13 by RealSenseCustomerSupport

pasquale90, 2021-07-29:
Open the **git** terminal and write:
```
cd <Anaconda installation path>/envs/<environment_name>
./Scripts/pip install pyrealsense2
```

After having install, check the installation by using:
```
conda activate <environment_name>
conda list
```
You will see pyrealsense2 in the list of installed packages within current environment